realsenseai/librealsense #2782: Data from RS2_STREAM_INFRARED and RS2_STREAM_DEPTH does not align

jasjuang, 2018-11-28:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.10.6 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC |
| SDK Version                            | 2.16.5 | 
| Language                            |  C++                         | 

### Issue Description
I recorded a depth image stream and a color image stream. The problem is they are not aligned with each other, and I couldn't find any information about the transformation between the color camera and the infrared cameras anywhere. Therefore I am trying to calibrate the infrared cameras and the color camera myself. When I look into rs_sensor.h, weirdly enough there is only one enum RS2_STREAM_INFRARED that represents the stream for an infrared camera but there is actually two infrared cameras. Unfortunately the stream from RS2_STREAM_INFRARED does not align with the stream from RS2_STREAM_DEPTH. I am guessing the stream from RS2_STREAM_DEPTH aligns with the other infrared camera that I don't have access to. 

How can I get access to stream of the other infrared camera?

Is there a way to directly record a depth stream that align with the color image stream?

Is there a way to directly obtain the transformation between the color camera and the infrared camera that aligns with the depth stream?

jasjuang, 2018-11-28:
I discovered the functions to obtain intrinsics and transformation between the color stream and the depth stream so this problem is not as urgent to me anymore, but I am still curious to know why the stream between RS2_STREAM_INFRARED and RS2_STREAM_DEPTH not align.

HippoEug, 2018-11-28:
From my understanding, all 3 streams (RGB, Infrared & Depth) are independent and all have different FOVs. To align them, you need a few other parameters as mentioned above

jasjuang, 2018-11-28:
Isn't the depth directly calculated from the stereo infrared camera with disparity estimation techniques? If that's the case there should be one depth map correspond to one infrared camera. Understand that only one of the depth map will be computed for speed purposes, but both infrared cameras should be accessible.

HippoEug, 2018-11-28:
"The Y16 format shows only unrectified images and is to be used only when calibrating the camera. Do not use Y16 format in your applications.

Use Y8 format in your applications. The Y8 format provides rectified images. This means that the depth images captured by each depth sensor have been rotated to align with each other so that the rotation matrix in the extrinsics is the identity matrix."

[Source](https://communities.intel.com/thread/126213)

dorodnic, 2018-11-28:
Hi @jasjuang 
Could you please share an image? 
Depth and infrared should be pixel-perfect aligned. Y16 is before rectification, but as far as I remember you can't stream it together with depth, so I'd like to better understand what you are seeing

jasjuang, 2018-11-28:
@dorian3d Please see the below for the infrared image (Y8) and depth image (Z16)

[ir_and_depth.zip](https://github.com/IntelRealSense/librealsense/files/2626433/ir_and_depth.zip)

As you can see they are not aligned.

dorodnic, 2018-11-29:
I think they are, and the depth image is just noisy. Can you try with `High Accuracy` preset?

closed 2018-12-10 by RealSense-Customer-Engineering