realsenseai/librealsense #2785: one channel depth image

SunshineDou, 2018-11-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello,
according to #1831 that I can get "Black to White"depth image now, but the depth image is still 24 bit, that's to say, it has three channels, but I want to get one channel 16 bit depth image. what should I do?

dorodnic, 2018-11-28:
Well, the original `depth_frame` is one channel 16 bit. Are you concerned with saturating the lower bits? 
Alternatively, you can use `opencv` to convert `colorizer` output to grayscale

SunshineDou, 2018-11-28:
thank you very much for your replying. I just want to get the original one channel 16 bit depth_frame to make datasets for 3D reconstruction "elasticfusion". but now I can only get three channel 24bit depth_frame. how can I get  original one channel 16 bit depth_frame?
Than you again for your help

dorodnic, 2018-11-28:
Can you give more info please. I assume C++? Do you need to create a CV object from it? 
The basic is:
```
pipeline p;
p.start();
auto fs = p.wait_for_frames();
auto df = fs.get_depth_frame();
uint16_t* ptr = (uint16_t*)df.get_data(); // 16-bpp depth data, with 10 effective bits in every pixel
```

SunshineDou, 2018-12-04:
@dorodnic Thank you very much for your replying
I just want to get aligned 16 bit depth image and color image to make datasets for 3D reconstruction
the code is as follows:
`import pyrealsense2 as rs
import numpy as np
import cv2
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

profile = pipeline.start(config)
depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

clipping_distance_in_meters = 1 #1 meter
clipping_distance = clipping_distance_in_meters / depth_scale

align_to = rs.stream.color
align = rs.align(align_to)
i=1
j=1
try:
    while True:
        frames = pipeline.wait_for_frames()
       
        aligned_frames = align.process(frames)
        
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = aligned_frames.get_color_frame()
        
        if not aligned_depth_frame or not color_frame:
            continue
        
        depth_image = (uint16_t*)(np.asanyarray(aligned_depth_frame.get_data()))
        color_image = np.asanyarray(color_frame.get_data())
        grey_color = 153
        depth_image_3d = np.dstack((depth_image,depth_image,depth_image)) #depth image is 1 channel, color is 3 channels
        bg_removed = np.where((depth_image_3d > clipping_distance) | (depth_image_3d <= 0), grey_color, color_image)
        
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        images = np.hstack((bg_removed, depth_colormap))
        cv2.namedWindow('Align Example', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Align Example', images)
        cv2.imwrite("//home//cindy//桌面//LabTest1//lab_depth//"+str(i)+".png",depth_image)  #保存到指定目录
        cv2.imwrite("//home//cindy//桌面//LabTest1//lab_color//"+str(i)+".png",color_image) 
        i = i+1
        j = j+1
        key = cv2.waitKey(1)
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
finally:
    pipeline.stop()
`
but it failed, what's the problem? thank you very much for your help

RealSense-Customer-Engineering, 2018-12-05:
[Realsense Customer Engineering Team Comment]
Hi @SunshineDou,

What do you mean "fail"? any error message? Is the pyrealsense library there?

SunshineDou, 2018-12-06:
@RealSense-Customer-Engineering 
Oh, I have solved it, It can get 16bit depth image now. Thank you very much sincerely~if I have any other question, I will open another issue

closed 2018-12-07 by RealSense-Customer-Engineering

lichenyue1996, 2019-11-20:
> @RealSense-Customer-Engineering
> Oh, I have solved it, It can get 16bit depth image now. Thank you very much sincerely~if I have any other question, I will open another issue

hello, I want to do the same things like you. I want to use realsense to save pictures like TUM dataset.
Could you tell me how to save depth images?