realsenseai/librealsense #2786: Align color image to depth using Labview wrapper

Metomotion, 2018-11-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Metomotion, 2018-11-28:
I am using the D435 with Labview wrapper. I got color image and depth, How can I align them using Labview wrapper?

dorodnic, 2018-11-28:
I think that alignment is not part of the provided example, but this can be done using C API. The high level flow would be:
1. Call `rs2_create_align` to get a pointer to a new `align` processing block. Pass the desired target stream type to the function.
2. Call `rs2_create_frame_queue` to create new frame queue for the aligned frames, pass 1 and NULL as parameters.
3. Bind align output to the queue using `rs2_start_processing_queue`. Pass the pointer from create_align and the queue.
4. Given a `frameset` object received either from `pipeline.wait_for_frames` or `syncer.wait_for_frames`, call `rs2_process_frame` on it.
5. Fetch the aligned frameset from the queue using `rs2_wait_for_frame` on the queue. 

Item #4 can be a bit tricky with existing labview layout, but if you want to create a new syncer, it can be done in very similar way (I think there is one in the project): 
1. `rs2_create_sync_processing_block`
2. `rs2_create_frame_queue`
3. `rs2_start_processing_queue`
4. `rs2_process_frame` on the color and `rs2_process_frame` on the depth frame
5. `rs2_wait_for_frame` to fetch the frameset

closed 2018-12-31 by RealSense-Customer-Engineering

tripleeagle10000, 2019-05-09:
I tried following the directions above but couldn't get it to work.  I have the RGB and Depth cameras streaming in LV but can't get the align processing block to do anything. 

Has anyone developed a LV example for this?  Some of the functions referenced in the directions above don't seem to have an equivalent LV VI (for example I can't find a way to use the pipeline, syncer, or use framesets in LV).  

Any help is appreciated.  Thanks

tripleeagle10000, 2019-06-05:
Just following up on the last post.  Anyone have any success aligning color and depth frames using Labview?

jo-wur, 2020-06-09:
I have the same question. Any success aligning color and depth frames of the D435 using Labview in the meantime?

wangzhuanxing123, 2022-03-24:
I have the same problem, has anyone solved it?

wangzhuanxing123, 2022-04-01:
After many days of hard work, I think I solved this problem. Intel's official example for LabVIEW has problems. In fact, it is not fully applicable to D435, and the content is not as rich as other programming languages. Therefore, in terms of RGB-Depth alignment, the example is applicable The RGB camera and the left camera (binocular camera) are Realsense devices with the same camera. They are not suitable for D435 and D435i. We need to obtain them through coordinate system transformation: First, we need to obtain the internal and external parameters of the camera. This is done after the SDK is installed. It is obtained from rs-sensor-control.exe in the tool folder of . Coordinates, then obtain RGB camera coordinates, and then obtain RGB pixel coordinates, which realizes the pixel conversion from depth map to RGB map. To convert RGB pixel coordinates to depth map coordinates, a corresponding approximate search is required
[image]
[image]
.

wangzhuanxing123, 2022-10-11:
您好，邮件已收到！
                     =--.赵东哲.--=