realsenseai/librealsense #2787: Questions about accessing other project scripts in Unity

GameHuiTea, 2018-11-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
Camera Model | { D415 }
Firmware Version | (Open RealSense Viewer --> Click info)
Operating System & Version | {Win (10)
Kernel Version (Linux Only) |  
Platform | PC
SDK Version | { 2.0 (build 2.16.5) }
Language | {unity }
Segment | { }

### Issue Description
<Describe your issue / question / feature request / etc..>
[image]

hello, have a good day!
I want to access the red square box script from the script in the yellow square box. but I can't access. 
The script in the yellow square box does not find the class in the red square box.
In a red square box, I can access the classes in a yellow square box, but not the reverse.

What should i do?
thank you for reading!

GameHuiTea, 2018-11-29:
@dorodnic Do you have any keywords for this problem?? I have no idea TAT

dorodnic, 2018-11-29:
Hi @GameHuiTea 
I'm not sufficiently familiar with the Unity wrapper, perhaps @ogoshen or @matkatz could comment?

ogoshen, 2018-11-29:
Red & Yellow are actually two different assemblies.
See Unity's docs about [ScriptCompilationAssemblyDefinitionFiles](https://docs.unity3d.com/Manual/ScriptCompilationAssemblyDefinitionFiles.html), and how to add references in the inspector.

Now it is my understanding that you are trying to add functionality to the RealSense components from external scripts which are separated on purpose. I suggest you keep them decoupled and redesign your components.

If you could post snippets of what you're trying to accomplish we might be able to help better.

HippoEug, 2018-11-30:
I am not too sure, but take a look at my [simple question/issue](https://github.com/IntelRealSense/librealsense/issues/2700). See the **Goal** part.

Maybe you should post an example of what you are trying to accomplish by modifying the Realsense code.

GameHuiTea, 2018-11-30:
I'm a little ignorant, but I found a way!
I made a new script and used it to copy the code of the script in the yellow square box.
It's working without a problem at the moment.

Thank you all for your answers!!

closed 2018-11-30 by GameHuiTea