realsenseai/librealsense #2789: Only few times can my computer recognize Realsense D435 as a USB 3.2

Medo91code, 2018-11-28:
Hi

Only few times can my computer recognize Realsense D435 as a USB 3.2 . Most of cases or times it recognize the camera only as USB 3.0 (USB 3.0 does not support the resolution 1080x1920 for RGB camera).

what i should do to make my computer , that it always recognize the camera as USB 3.2 ?

Thank you
[image]
[image]

dorodnic, 2018-11-28:
Hi @Medo91code 
This question is rather specific to the type of USB controller and cable you are using. I personally don't have a good answer for the general case.
What resolutions do you see when you expand the "RGB Camera" block in the 3.0 case? Is 1080x1920 missing from the list?

Medo91code, 2018-11-28:
> This question is rather specific to the type of USB controller and cable you are using

I use USB 3.0 Port and the original cable (cable was in the Box ) .
the Controller is : intel usb 3.0 extensible host controller.

[image]

> What resolutions do you see when you expand the "RGB Camera" block in the 3.0 case? Is 1080x1920 missing from the list?

yes , in USB 3.0 case the resolution1080x1920 is missing from the list.
only it is available in USB 3.2 case.

thank you

RealSense-Customer-Engineering, 2018-11-30:
[Realsense Customer Engineering Team Comment]
Hi

Is this issue concluded that it only happened on your specific computer? I have tried my D435 on Windows 8.1 x64, but couldn't reproduce this issue. Maybe I have different build version.

Which side of cable do you plug/unplug? Host side or Device side? (or say on Type A end or Type C end?) If plug/unplug on Type C end, do you seen same behavior after flipping Type C plug? Have you tried other USB-IF certified cable (with SS logo)?

Thanks

Medo91code, 2018-12-05:
> is this issue concluded that it only happened on your specific computer? ``

Yes , only it is happened on my computer

> Which side of cable do you plug/unplug? Host side or Device side? (or say on Type A end or Type C end?) If plug/unplug on Type C end, do you seen same behavior after flipping Type C plug? Have you tried other USB-IF certified cable (with SS logo)?

i try to plug/unplug both Type A and  Type C , but nothing happened . sometimes (not always) my computer can recognize USB 3.2  when i restart it .

Thanks

RealSense-Customer-Engineering, 2018-12-05:
[Realsense Customer Engineering Team Comment]
Sorry, would you like to take a chance to reinstall the Windows OS ?

So-LiYatao, 2018-12-06:
I also encountered the same problem on my laptop, sometimes inexplicably became USB3.0, unable to use Visual Studio debugging camera, but sometimes restore USB3.2. I tried to upgrade win10 system, but still will encounter the same problem.I have the same problem on another computer, but the chances are slim.  I don't know how to solve this problem.

RealSense-Customer-Engineering, 2018-12-06:
[Realsense Customer Engineering Team Comment]
Can you let us know more details about your laptop? the chipset info and Windows OS info.

Thanks

So-LiYatao, 2018-12-07:
Hello, my laptop is the previous one. The model is asus K550V, CPU i53230M, main frequency 2.6GHz, and main board chipset is Intel HM76. Win10 version is 1809 (OS internal version 17763.107)

Medo91code, 2018-12-08:
information about my laptop 
[image]

RealSense-Customer-Engineering, 2018-12-10:
[Realsense Customer Engineering Team Comment]
Hi

I am trying to locate the similar platform with same chipset.

Meantime, do you have a chance to find a self-powered external USB3 hub (USB-IF certified) and give it a shot?

Thanks

RealSense-Customer-Engineering, 2019-01-10:
[Realsense Customer Engineering Team Comment]
Hi

Sorry, I couldn't locate same platform as yours here. Have you tried using a self-powered USB3 hub (USB-IF certified) ?
Thanks

closed 2019-05-12 by dorodnic