realsenseai/librealsense #2792: Kernel patching on Xavier

pavloblindnology, 2018-11-28:
What are your plans to release kernel patching scripts for Xavier (Linux Kernel 4.9).
Thanks.

ev-mp, 2018-11-28:
Hello @pavloblindnology, 
There are no immediate plans for Kernel 4.9/Xavier and we continue to maintain and support Ubuntu LTS kernels line. Note that Librealsense already supports [NVidia's Jetson-TX2](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md).

We do periodic reviews and gradually expand the support to additional platforms and distributions, taking community inputs as well, so community-pushed patches ports would most certainly be accepted.

In the short term we are also going to publish kernels support explanatory to present the pros and cons of the kernel modifications and the expected results.

pavloblindnology, 2018-11-28:
Is TX2 installation procedure going to work for Xavier?

dorodnic, 2018-11-28:
Hi @pavloblindnology 

We do not officially validate RealSense products on non-x86 platforms, so we cannot guaranty that it will work out of the box.
There is a CMake option that allows you to bypass most kernel patching, `-DFORCE_LIBUVC=true`
Also the team has sufficient experience to assist you with porting the patches, if you choose to give it a try.

See also [#2676](https://github.com/IntelRealSense/librealsense/issues/2676#issuecomment-436147539)

VasuAgrawal, 2018-11-28:
Hi @pavloblindnology, @dorodnic 

I was able to get the kernel modules patched to include hardware timestamping on my Xavier, running kernel 4.9. I had to build from source and make some changes to some of the patches, but hardware timestamping (and all the other metadata) works for us. I didn't patch the UVC driver to include all of the different streaming formats since RGB8 and Y16 suit our needs.

There's some (poorly written) documentation here:

https://docs.google.com/document/d/1vYK9-Jz0PKkNzqu5F_ARxl9zsEAdTOQs9ZdAL-Clobk/edit?usp=sharing

If anything's unclear, let me know and I can try to improve it. I did my best to document everything as I went along, but there's a chance I missed stuff.

pavloblindnology, 2018-12-13:
I currently work on creation of a patching script for Xavier following `patch-realsense-ubuntu-lts.sh` script and PR #1438 .
I've managed to modify the patches
`realsense-camera-formats_ubuntu-xenial-hwe-zesty.patch`
`realsense-metadata-ubuntu-xenial-hwe-zesty.patch`
appropriately and use original
`realsense-hid-ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1.patch`
`realsense-powerlinefrequency-control-fix.patch`
`v1-media-uvcvideo-mark-buffer-error-where-overflow.patch` (is this one needed?)
to patch successfully.
Then I use [configureKernel.sh](https://github.com/jetsonhacks/buildLibrealsense2TX/blob/6ad3ef0cc286e4f4c8c0637e3ae019c7eb382f53/scripts/configureKernel.sh#L28) script from [jetsonhacks/buildLibrealsense2TX](https://github.com/jetsonhacks/buildLibrealsense2TX) to turn on accel and gyro external modules build.
After modules' building process I get
`uvcvideo.ko, hid-sensor-accel-3d.ko, hid-sensor-gyro-3d.ko, videobuf-core.ko, videobuf-vmalloc.ko` modules, but no `videodev.ko` module.
I've tried to set `VIDEO_DEV` and `VIDEO_V4L2` to `m` instead of `y`,
but with no luck - they have been reverted to `y` after `make oldconfig`.

`uvcvideo.ko, videobuf-core.ko, videobuf-vmalloc.ko` modules are installed successfully.
With `hid-sensor-accel-3d.ko, hid-sensor-gyro-3d.ko` modules I get error `modprobe: FATAL: Module hid-sensor-accel-3d not found in directory /lib/modules/4.9.108-tegra`

What am I doing wrong and what other suggestions does someone have?
Thanks.

RealSenseCustomerSupport, 2019-01-22:
Hi @pavloblindnology,
Were you be able to overcome the accel and gyro build issue?

pavloblindnology, 2019-01-23:
@RealSenseCustomerSupport Nope, I asked on Nvidia forum but was redirected here. So I stopped proceeding with that.

MartyG-RealSense, 2019-02-19:
As this case is still open, I just wanted to add that JetsonHacks have published a guide to setting up Librealsense2 on the Xavier.

https://www.jetsonhacks.com/2019/01/21/intel-realsense-d435i-on-nvidia-jetson-agx-xavier/

RealSenseCustomerSupport, 2019-03-02:
----------------------------------------------

Thank you very much @MartyG-RealSense and also for making announcement for Guide to Enable the D435i on NVIDIA Jetson AGX Xavier:
https://github.com/IntelRealSense/librealsense/issues/3290
Closing this ticket for now.

closed 2019-03-09 by RealSenseCustomerSupport