realsenseai/librealsense #2795: Cannot set rs.option.depth_units

xkunwu, 2018-11-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |  Ubuntu 18.04   | 
| Kernel Version (Linux Only)    |  4.15.0-39                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2                          | 
| Language                            |  python                          | 
| Segment			|  others                          | 

### Issue Description
This is a following up to [Issue#1905](https://github.com/IntelRealSense/librealsense/issues/1905#issuecomment-442532214), in which changing depth scale was discussed.

The code:
`        depth_sensor = profile.get_device().first_depth_sensor()
        depth_sensor.set_option(rs.option.depth_units, 0.001)
`
Error:
`RuntimeError: This option is read-only!`

dorodnic, 2018-11-29:
Hi @xkunwu 
Changing depth units is not possible with the SR300, only with the D400 series of cameras.

xkunwu, 2018-11-29:
Hi @dorodnic 
Thanks! Good to know ... I am closing this issue.

closed 2018-11-29 by xkunwu