realsenseai/librealsense #2796: How to convert realsense RGB into byte []?

laolaolulu, 2018-11-29:
[image]

dorodnic, 2018-11-29:
Can this help: [cs-tutorial-1-depth/Program.cs:43-50](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/cs-tutorial-1-depth/Program.cs#L43-L50)

laolaolulu, 2018-11-29:
[image]

ogoshen, 2018-11-29:
You are trying to copy from a color frame to incompatible `ushort[]` depth data, causing a buffer overflow.

I'm assuming you meant to copy from `frames.DepthFrame`
If it is the color data you want, the buffer should look like:
```cs
byte[] color = new byte[1920*1080*3] // assuming 1080 RGB8
```

closed 2018-12-03 by laolaolulu