realsenseai/librealsense #2801: missing:  REALSENSE_LIBRARY REALSENSE_INCLUDE_DIR

SunshineDou, 2018-11-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
hello, I want to compile a project on Ubuntu16, but when I cmake, the result as follows:
missing:  REALSENSE_LIBRARY REALSENSE_INCLUDE_DIR
by cmake-gui, I changed my realsense path to usr/local/include/librealsense2 and usr/include/librealsense2, but it still failed as above :missing:  REALSENSE_LIBRARY REALSENSE_INCLUDE_DIR
can you help me? what's the problem? why it cannot find realsense library and realsense include dir ?

dorodnic, 2018-11-29:
Hi @SunshineDou 
We recently added [cmake example](https://github.com/IntelRealSense/librealsense/tree/development/examples/cmake), maybe it can help.
It should work after either building SDK from source and doing `sudo make install` or after installing from apt-get.

ev-mp, 2018-11-29:
@SunshineDou , you haven't specified the version you're running, but in the recent v2.16.5 some CMake variables were modified, including:
`realsense_INCLUDE_DIR -> realsense2_INCLUDE_DIR`
See the [release notes](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2165).

This can affect affect you as well

SunshineDou, 2018-11-30:
Thanks for your generously help

RealSense-Customer-Engineering, 2018-12-12:
[Realsense Customer Engineering Team Comment]
Hi @SunshineDou,

still need any support for this topic?

SunshineDou, 2018-12-13:
@RealSense-Customer-Engineering 
I have no question about this problem now. if I have any other problem, I will open another issue. I will close this issue. Thank you very much

closed 2018-12-13 by SunshineDou