realsenseai/librealsense #2802: How to save depth images with opencv after post processing?

AIBluefisher, 2018-11-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.09.09.02 | 
| Operating System & Version |    Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.15.0-39-generic                                        | 
| Platform                                 | PC |
| SDK Version                            |  2                         | 
| Language                            |  C++                           | 
| Segment			| others                       | 

### Issue Description
Hi,
I want to store the depth images after post processing with OpenCV. I can save the right images before post processing, however, save wrong images after post processing.
```C++
frames = pipe.wait_for_frames();
for (auto&& frame : pipe.wait_for_frames()) {
    if (auto vf = frame.as<rs2::video_frame>()) {
        std::stringstream png_file;
        std::string img_type = vf.get_profile().stream_name();
        std::transform(img_type.begin(), img_type.end(), img_type.begin(), ::tolower);
        png_file << outdir << "/frame-" << std::setw(6) << std::setfill('0') 
                     << std::to_string(frame_no) << "." << img_type  << ".png";
                  
        if (vf.is<rs2::depth_frame>()) {
            // post processing
            rs2::frame filtered = vf;
            if (filter_enabled) {
                rs2::decimation_filter dec_filter;
                rs2::spatial_filter spat_filter;
                rs2::temporal_filter temp_filter;
                rs2::disparity_transform depth_to_disparity(true);
                rs2::disparity_transform disparity_to_depth(false);
                filtered = dec_filter.process(filtered);
                filtered = depth_to_disparity.process(filtered);
                filtered = spat_filter.process(filtered);
                filtered = temp_filter.process(filtered);
                filtered = disparity_to_depth.process(filtered);
            
            // Use the colorizer to get an rgb image for the depth stream
            // auto vf = color_map.process(frame);
            auto vf = color_map.process(filtered);
            std::cout << "Format: " << vf.get_profile().format() << std::endl;
            depth_image.render(vf, { app.width() / 2, 0, app.width() / 2, app.height() });
            Mat img(Size(imgSize.x, imgSize.y), CV_8UC3, (void*)vf.get_data(), Mat::AUTO_STEP);
            SaveImage(img, png_file.str());
        }
        else {
            color_image.render(vf, { 0,               0, app.width() / 2, app.height() });
            Mat img(Size(imgSize.x, imgSize.y), CV_8UC3, (void*)vf.get_data(), Mat::AUTO_STEP);
            Mat bgr_img;
            cvtColor(img, bgr_img, CV_RGB2BGR);
            SaveImage(bgr_img, png_file.str());
        }
        std::cout << "Saved " << png_file.str() << std::endl;
    }
}          
......

void SaveImage(Mat img, string path)
{
	vector<int> compression_params;
	compression_params.push_back(IMWRITE_PNG_COMPRESSION);
	compression_params.push_back(9);
	// compression_params.push_back(0);
        // compression_params.push_back(IMWRITE_PNG_STRATEGY);
        // compression_params.push_back(IMWRITE_PNG_STRATEGY_DEFAULT);
	try {
		imwrite(path, img, compression_params);
	}
	catch (cv::Exception& ex) {
		cout << "Exception converting image to PNG format: %s\n" << ex.what();
		return;
	}
}
```
The image **before post processing** and **after post processing** is respectively shown below:
[image]
[image]

Both the frame format before and after post processing are **RGB8**. I have no idea what's wrong.
 Can you help me? Thanks so much!

ev-mp, 2018-12-02:
This line look suspicious:
> Mat img(Size(**imgSize.x, imgSize.y**), CV_8UC3, (void*)vf.get_data(), Mat::AUTO_STEP);

The decimation filter [modifies the image size](https://github.com/IntelRealSense/librealsense/blob/c3c758d18c585a237bb5b635927797aa69996391/doc/post-processing-filters.md#decimation-filter)
>filtered = dec_filter.process(filtered);

But the modified values don't seem to be accounted for in the code

AIBluefisher, 2018-12-03:
Got it, Thank you very much! But is there any way that don't modify the image size with post processing?

So-LiYatao, 2018-12-07:
Hello, I'm from zhejiang, China.I am using a D435 camera, and I want to get the post-processed image on OpenCV. Similar to you, I have checked the rs_post processing. CPP in SDK, but I have not solved the problem of obtaining the processed image in OpenCV.Hope to learn from your code.

AIBluefisher, 2018-12-09:
> Absolutely.
> The post-processing filters (except for depth<->disparity) are not coupled, therefore can be reordered/skipped/invoked multiple times and so forth.
> 
> Just comment out the call to decimation as it is the only filter that modifies the image size.
> Note that w/o decimation filter above , the rest of filters will have 4 times more pixels to process which may impact performance.

It really helps me a lot. Thank you very much!

closed 2018-12-09 by AIBluefisher