realsenseai/librealsense #2806: Unknown symbol in videodev, uvcvideo errors

jeigrei, 2018-11-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 2.17.0 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-34                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2                     | 
| Language                            |  -                       | 
| Segment			|  - | 
### Issue Description

After following the instructions to install from debian packages (https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md), trying to run the realsense-viewer yields
```
(backend-v4l2.cpp:518) Cannot access /sys/class/video4linux
```
watching with `dmesg` shows the following when I plug in the RealSense camera:

```
[Thu Nov 29 18:27:44 2018] usb 4-6: new SuperSpeed USB device number 13 using xhci_hcd
[Thu Nov 29 18:27:44 2018] usb 4-6: New USB device found, idVendor=8086, idProduct=0b07
[Thu Nov 29 18:27:44 2018] usb 4-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Nov 29 18:27:44 2018] usb 4-6: Product: Intel(R) RealSense(TM) 430
[Thu Nov 29 18:27:44 2018] usb 4-6: Manufacturer: Intel(R) RealSense(TM) 430
[Thu Nov 29 18:27:44 2018] usb 4-6: SerialNumber: 822213021991
[Thu Nov 29 18:27:44 2018] videodev: Unknown symbol media_devnode_remove (err 0)
[Thu Nov 29 18:27:44 2018] videodev: Unknown symbol media_entity_put (err 0)
[Thu Nov 29 18:27:44 2018] videodev: Unknown symbol media_create_intf_link (err 0)
...
```
and so on for 60 lines or so. The same errors get shown in dmesg from a `sudo modprobe uvcvideo`, which returns 
```
modprobe: ERROR: could not insert 'uvcvideo': Unknown symbol in module, or unknown parameter (see dmesg)
```
Looking at both the uvcvideo and videodev modules with `modinfo` shows them apparently using the correct version:
```
> modinfo uvcvideo      
filename:       /lib/modules/4.15.0-34-generic/updates/dkms/uvcvideo.ko
version:        1.1.2.realsense-1.3.4
...
vermagic:       4.15.0-34-generic SMP mod_unload 
> modinfo videodev
filename:       /lib/modules/4.15.0-34-generic/updates/dkms/videodev.ko
version:        realsense2-dkms
...
vermagic:       4.15.0-34-generic SMP mod_unload 
```

Any guidance here would be much appreciated, as I'm a little lost with finding what the problem might be

dorodnic, 2018-11-30:
Hi @jeigrei 
Could you please share `dkms status`?
This can happen if RealSense DKMS is conflicting with another DKMS installed on the system. 
Specifically, this once happened to me with older NVidia drivers. For me the issue was resolved by carefully uninstalling both, reinstalling RealSense DKMS and then installing the latest graphics drivers.

jeigrei, 2018-11-30:
Sure, `dkms status` yields

```
librealsense2-dkms, 1.3.4, 4.15.0-34-generic, x86_64: installed
librealsense2-dkms, 1.3.4, 4.15.0-39-generic, x86_64: installed
librealsense2-dkms, 1.3.4, 4.4.0-139-generic, x86_64: installed
virtualbox, 5.1.34, 4.13.0-41-generic, x86_64: installed
virtualbox, 5.1.34, 4.13.0-43-generic, x86_64: installed
virtualbox, 5.1.34, 4.13.0-45-generic, x86_64: installed
virtualbox, 5.1.34, 4.15.0-24-generic, x86_64: installed
virtualbox, 5.1.34, 4.15.0-29-generic, x86_64: installed
virtualbox, 5.1.34, 4.15.0-30-generic, x86_64: installed
virtualbox, 5.1.34, 4.15.0-32-generic, x86_64: installed
virtualbox, 5.1.34, 4.15.0-33-generic, x86_64: installed
virtualbox, 5.1.34, 4.15.0-34-generic, x86_64: installed
virtualbox, 5.1.34, 4.15.0-36-generic, x86_64: installed
virtualbox, 5.1.34, 4.15.0-39-generic, x86_64: installed
virtualbox, 5.1.34, 4.4.0-139-generic, x86_64: installed
```

jeigrei, 2018-11-30:
So I tried uninstalling both DKMS and reinstalling RealSense, just to make sure virtualbox wasn't causing any conflicts:
```
> sudo apt-get purge virtualbox-\*
> sudo apt-get remove librealsense2-dkms
> sudo apt-get install librealsense2-dkms
> dkms status
librealsense2-dkms, 1.3.4, 4.15.0-34-generic, x86_64: installed  
librealsense2-dkms, 1.3.4, 4.15.0-39-generic, x86_64: installed
```
Didn't work -- seeing the same errors as in the original post

jeigrei, 2018-11-30:
#### Update
Still not sure what the problem was, but I was able to resolve this by uninstalling the dkms for the kernel I'm _not_ using, i.e.
```
sudo dkms uninstall -k 4.15.0-39-generic librealsense2-dkms/1.3.4
```
and then restarting after that.

closed 2018-11-30 by jeigrei

pangchenglin, 2019-06-19:
I have the same problem 
`> dkms status`
`librealsense2-dkms, 1.3.5, 4.15.0-51-generic, x86_64: installed
`
Can you help me?

adamconkey, 2023-09-07:
I was seeing the same behavior on Ubuntu 22.04 with kernel `5.15.0-83`, I tried many things but the solution ended up coming from [the linked issue above](https://github.com/IntelRealSense/librealsense/issues/6627), where I just did
```bash
sudo apt install linux-modules-extra-$(uname -r)
```
and rebooted, everything worked after that.