realsenseai/librealsense #2811: Installation SDK on the Mac

GitwellAnyohub, 2018-11-30:
I've been working with the SDK on linux for some time now and I've written an idiot proof bash script to install from both source and also from building.

I now need to work on a Mac and the installation guide is extremely sparse. My expectation from any installation guide is that I can copy and paste the code line by line into the terminal and the installation will work. However, this is clearly not the case for the guide below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

For one thing, none of these lines specify anything about the librealsense library. 

mkdir build && cd build
sudo xcode-select --reset
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -G Xcode

what is this?

Im supposed to make a directly called build that is completely empty and cmake and that's supposed to work? What am I "making"? Are there more detailed step by step instructions so a beginner that doesn't know what any of this is can get this up and running on a Mac? My assumption is that a ton of steps are missing.

dorodnic, 2018-11-30:
Hi @GitwellAnyohub 
You could try `brew install librealsense`

GitwellAnyohub, 2018-12-03:
I can actually get the realsense-viewer to work fine. What I need is to be able to import pyrealsense2 in python.

GitwellAnyohub, 2018-12-04:
For anyone that still needs to get the SDK working on Macs, here is what I did. It was trivial for me to get the command line stuff working (realsense-viewer) with brew install librealsense, but the finicky thing (for me) was to get the python library working. 

https://github.com/GitwellAnyohub/Real_Sensible_Moseq/blob/master/bash_and_install/install_realsense_sdk_mac_os.sh

dorodnic, 2018-12-05:
Thank you @GitwellAnyohub we will expand the documentation accordingly. 
Ideally, we want to publish `pip` packages for Mac, like we do for Windows and Linux, but it is taking some time.

flyingfalafel, 2020-02-13:
If it all goes without errors, you should be able to find the pyrealsense2.<arch info>.so under build/wrappers/python (actually 3 files with the same name and extensions .so, .so.2, .so.2.8.1).
Now the easiest way to use it is run python from that folder and import pyrealsense2

source: https://github.com/IntelRealSense/librealsense/issues/793

exnx, 2020-10-15:
Ridiculous to still not have updated the official install for MacOS...  Intel should be embarrassed with themselves. 

Why not just use the instructions @GitwellAnyohub posted at the least, that worked for me.

eweimholt, 2020-10-23:
@RealSenseSupport I'd like to echo @exnx and also say I am surprised that the macOS installation guide is still incomplete... I'm going to try @GitwellAnyohub's instructions now

Borzborz, 2020-11-04:
I am having the same trouble as @eweimholt and @exnx, I followed @GitwellAnyohub's instructions and everything worked fine until I tried to import the pyrealsense2 in python. I get a "ModuleNotFoundError:  No module named 'pyrealsense2'". Any help would be appreciated

jalapatip, 2021-06-29:
I tried installing librealsense using HomeBrew but am getting the following error. 

"Error: librealsense: no bottle available!"