realsenseai/librealsense #2813: Can I use the librealsense SDK in Qt (Mac OS)

Core055, 2018-12-01:
Camera Model : D435
OS : Mac OS
Language : C++   (with Qt)

 I want to  konw  how to install the project with Qt   in  Mac OS  like the following method?
[image]
thank you

Core055, 2018-12-05:
how to  solve it？？？？？

RealSense-Customer-Engineering, 2018-12-07:
[Realsense Customer Engineering Team Comment]
Hi @Core055,
After following the instruction: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

Please open and build the project with Qt by: From Qt Creator > Open File/Project > [librealsense/CMakeLists.txt]