realsenseai/librealsense #2825: OpenNI2 Driver doesn't works with NiTE2

UnaNancyOwen, 2018-12-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400                                                       | 
| Firmware Version                   | 05.10.13.00                                             | 
| Operating System & Version |  Windows 10                                          | 
| SDK Version                            |  v2.17.0                                                  | 
| Language                               |  OpenNI2/NiTE2                                     | 

### Issue Description
The OpenNI2 Driver (<code>rs2driver.dll</code>) works with OpenNI2, but doesn't works with NiTE2.
I tried OpenNI2 Driver with <code>OpenNI2/Tools/NiViewer.exe</code>. It works correctly.
Then, I tried OpenNI2 Driver with <code>NiTE2/Samples/SimpleUserTracker.exe</code>. It doesn't works.
This is the output of <code>SimpleUserTracker.exe</code>. It output error <code>Could not get shift2Depth table.</code>.

### Output SimpleUserTracker.exe
```
PS C:\Program Files\PrimeSense\NiTE2\Samples\Bin> .\SimpleUserTracker.exe
[RS2] +Rs2Driver
[RS2] oni::driver::Rs2Driver::initialize
[RS2] oni::driver::Rs2Driver::enumerateDevices
[RS2] oni::driver::Rs2Driver::devicesChanged removed=0000000000000000 added=0000000002B8D4B0
[RS2] deviceConnected
[RS2] oni::driver::Rs2Device::queryDeviceInfo
[RS2] DEVICE serial=746112061315 name=Intel RealSense D415
[RS2] Rs2Driver INITIALIZED
[RS2] oni::driver::Rs2Driver::deviceOpen uri=746112061315
[RS2] +Rs2Device
[RS2] oni::driver::Rs2Device::initialize
[RS2] oni::driver::Rs2Device::queryDeviceInfo
[RS2] DEVICE serial=746112061315 name=Intel RealSense D415
[RS2] oni::driver::Rs2Device::initializeStreams
[RS2] SENSOR 0
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=10 width=1920 height=1080 framerate=25
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=10 width=1920 height=1080 framerate=25
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=9 width=1920 height=1080 framerate=25
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=9 width=1920 height=1080 framerate=25
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=10 width=1920 height=1080 framerate=15
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=10 width=1920 height=1080 framerate=15
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=9 width=1920 height=1080 framerate=15
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=9 width=1920 height=1080 framerate=15
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=9 width=1280 height=720 framerate=30
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=9 width=1280 height=720 framerate=30
[RS2]   type=3 sensorId=0 streamId=0 profileId=5 format=5 width=1280 height=720 framerate=30
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=9 width=1280 height=720 framerate=15
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=9 width=1280 height=720 framerate=15
[RS2]   type=3 sensorId=0 streamId=0 profileId=5 format=5 width=1280 height=720 framerate=15
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=9 width=1280 height=720 framerate=6
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=9 width=1280 height=720 framerate=6
[RS2]   type=3 sensorId=0 streamId=0 profileId=5 format=5 width=1280 height=720 framerate=6
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=10 width=960 height=540 framerate=25
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=10 width=960 height=540 framerate=25
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=10 width=960 height=540 framerate=15
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=10 width=960 height=540 framerate=15
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=9 width=848 height=480 framerate=90
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=9 width=848 height=480 framerate=90
[RS2]   type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=90
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=9 width=848 height=480 framerate=60
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=9 width=848 height=480 framerate=60
[RS2]   type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=60
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=9 width=848 height=480 framerate=30
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=9 width=848 height=480 framerate=30
[RS2]   type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=30
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=9 width=848 height=480 framerate=15
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=9 width=848 height=480 framerate=15
[RS2]   type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=15
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=9 width=848 height=480 framerate=6
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=9 width=848 height=480 framerate=6
[RS2]   type=3 sensorId=0 streamId=0 profileId=5 format=5 width=848 height=480 framerate=6
[RS2]   type=3 sensorId=0 streamId=1 profileId=5 format=9 width=640 height=480 framerate=90
[RS2]   type=3 sensorId=0 streamId=2 profileId=6 format=9 width=640 height=480 framerate=90
[... 383 lines cut ...]
[RS2]   type=2 sensorId=1 streamId=0 profileId=7 format=4 width=320 height=240 framerate=30
[RS2]   type=2 sensorId=1 streamId=0 profileId=7 format=5 width=320 height=240 framerate=6
[RS2]   type=2 sensorId=1 streamId=0 profileId=7 format=10 width=320 height=240 framerate=6
[RS2]   type=2 sensorId=1 streamId=0 profileId=7 format=4 width=320 height=240 framerate=6
[RS2]   type=2 sensorId=1 streamId=0 profileId=7 format=5 width=320 height=180 framerate=60
[RS2]   type=2 sensorId=1 streamId=0 profileId=7 format=10 width=320 height=180 framerate=60
[RS2]   type=2 sensorId=1 streamId=0 profileId=7 format=4 width=320 height=180 framerate=60
[RS2]   type=2 sensorId=1 streamId=0 profileId=7 format=5 width=320 height=180 framerate=30
[RS2]   type=2 sensorId=1 streamId=0 profileId=7 format=10 width=320 height=180 framerate=30
[RS2]   type=2 sensorId=1 streamId=0 profileId=7 format=4 width=320 height=180 framerate=30
[RS2]   type=2 sensorId=1 streamId=0 profileId=7 format=5 width=320 height=180 framerate=6
[RS2]   type=2 sensorId=1 streamId=0 profileId=7 format=10 width=320 height=180 framerate=6
[RS2]   type=2 sensorId=1 streamId=0 profileId=7 format=4 width=320 height=180 framerate=6
[RS2] oni::driver::Rs2Device::mainLoop
[RS2] oni::driver::Rs2Device::getSensorInfoList
[RS2] oni::driver::Rs2Device::createStream sensorType=3
[RS2] oni::driver::Rs2Stream::start type=1 sensorId=0 streamId=0
[RS2] oni::driver::Rs2Device::updateConfiguration
Couldn't get shift2Depth table.
[RS2] oni::driver::Rs2Device::restartPipeline
[RS2] oni::driver::Rs2Device::startPipeline
[RS2] rs2_create_config
[RS2] ENABLE STREAM type=1 sensorId=0 streamId=0 640x480 @30
[RS2] rs2_create_pipeline
```

wongfei, 2018-12-02:
It's possible but you need to recompile driver with patch:
https://gist.github.com/wongfei/104e5b382f82295deed0f38a07e9a7f3

Brekel, 2018-12-03:
After applying the patch, it seems in the OpenNI NiTE UserViewer project, in the Viewer.cpp file, in the DrawLimb function:

Joint coordinates have invalid/infinite values, they are a result of pUserTracker->convertJointCoordinatesToDepth (which in turn relies on niteConvertJointCoordinatesToDepth).
The joint confidences seem reasonable though.

wongfei, 2018-12-03:
For some reason convertJointCoordinatesToDepth doesn't work, hard to figure out why because NITE is binary only. But it can be replaced with rs2_project_point_to_pixel:

**Rs2Device.cpp**

```
#define RS2_PROJECT_POINT_TO_PIXEL 0x1000
struct Rs2PointPixel
{
	float point[3];
	float pixel[2];
};

OniStatus Rs2Device::invoke(int commandId, void* data, int dataSize)
{
	if (commandId == RS2_PROJECT_POINT_TO_PIXEL && data && dataSize == sizeof(Rs2PointPixel))
	{
		for (auto iter = m_createdStreams.begin(); iter != m_createdStreams.end(); ++iter)
		{
			Rs2Stream* stream = *iter;
			if (stream->getOniType() == ONI_SENSOR_DEPTH)
			{
				auto proj = (Rs2PointPixel*)data;
				rs2_project_point_to_pixel(proj->pixel, &stream->m_intrinsics, proj->point);
				return ONI_STATUS_OK;
			}
		}
	}
		
	return ONI_STATUS_NOT_SUPPORTED;
}

OniBool Rs2Device::isCommandSupported(int commandId)
{
	switch (commandId)
	{
		case RS2_PROJECT_POINT_TO_PIXEL: return true;
	}

	return false;
}
```

**Viewer.cpp**

```
#define RS2_PROJECT_POINT_TO_PIXEL 0x1000
struct Rs2PointPixel
{
	float point[3];
	float pixel[2];
};

void projectPointToPixel(float x, float y, float z, float* pOutX, float* pOutY)
{
	Rs2PointPixel proj;
	memset(&proj, 0, sizeof(proj));
	proj.point[0] = x;
	proj.point[1] = y;
	proj.point[2] = z;

	auto viewer = SampleViewer::GetInstance();
	viewer->GetDevice().invoke(RS2_PROJECT_POINT_TO_PIXEL, (void*)&proj, (int)sizeof(proj));

	*pOutX = proj.pixel[0];
	*pOutY = g_nYRes - proj.pixel[1];
}
```

**Viewer.h**

```
class SampleViewer
{
public:
	static SampleViewer* GetInstance() { return ms_self; }
	openni::Device& GetDevice() { return m_device; }
```

UnaNancyOwen, 2018-12-04:
Thanks, It is works correctly! 👍 
@wongfei Do you have a plan to send a pull request to fix rs2driver.dll? It's great.

**Viewer.cpp**
```patch
  void DrawLimb(nite::UserTracker* pUserTracker, const nite::SkeletonJoint& joint1, const nite::SkeletonJoint& joint2, int color)
  {
	float coordinates[6] = {0};
- 	pUserTracker->convertJointCoordinatesToDepth(joint1.getPosition().x, joint1.getPosition().y, joint1.getPosition().z, &coordinates[0], &coordinates[1]);
-  	pUserTracker->convertJointCoordinatesToDepth(joint2.getPosition().x, joint2.getPosition().y, joint2.getPosition().z, &coordinates[3], &coordinates[4]);
+	projectPointToPixel(joint1.getPosition().x, joint1.getPosition().y, joint1.getPosition().z, &coordinates[0], &coordinates[1]);
+	projectPointToPixel(joint2.getPosition().x, joint2.getPosition().y, joint2.getPosition().z, &coordinates[3], &coordinates[4]);
```

wongfei, 2018-12-06:
> Do you have a plan to send a pull request to fix rs2driver.dll?
Nope, but someone with license/legal knowledge can do it, because shift tables are reverse engineered (Kinect OpenNI driver also uses them).

UnaNancyOwen, 2018-12-07:
> someone with license/legal knowledge can do it,

I think it requires help by RealSense team.
@RealSense-Customer-Engineering @dorodnic What do you think?

UnaNancyOwen, 2018-12-13:
@RealSense-Customer-Engineering @dorodnic Please tell us your opinion. Thanks,

dorodnic, 2018-12-22:
Hi @UnaNancyOwen 
I apologize, I'm not 100% sure what is the correct answer, legal-wise. Could you (or @wongfei) please open a pull-request with the exact changes?
There is an automated Intel infrastructure in place that checks for license violations, nightly. If the change will trigger an error, we will have to roll-back the PR. My intuition is that it is Ok, for the reasons you stated above.

UnaNancyOwen, 2018-12-24:
> Could you (or @wongfei) please open a pull-request with the exact changes?

@wongfei Could you send a pull request? (If you want it, I can send pull request instead of you.)

RealSense-Customer-Engineering, 2019-01-08:
[Realsense Customer Engineering Team Comment]
Hi @UnaNancyOwen, @wongfei,

Is the pull request filed in order to fix this?

UnaNancyOwen, 2019-01-08:
> Is the pull request filed in order to fix this?

No yet, @wongfei What about you?

closed 2019-02-19 by RealSenseCustomerSupport