realsenseai/librealsense #2827: Align color to depth without empty black areas

KarlAbson, 2018-12-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 / D435 | 
| Firmware Version                   | (05.10.06) | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.16.5                        | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I'm currently playing with the rs-measure c++ example. 
I understand that the colour needs to be aligned to depth for the correct measurements to be given.
I notice however that during the alignment process that this adds these "black areas". This is where it has no depth information I believe? Is it possible to align without these black areas? From a user perspective it would be much nicer looking.

dorodnic, 2018-12-03:
Hi @KarlAbson 
We have couple of hole-filling capabilities within our post-processing suite. I have a [draft notebook](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb) illustrating the effect of different filters.
The source of black pixels in the aligned image is not just pixels without depth. 
Anytime you align from one sensor to another, there are going to be pixels that one sensor can see but the other cannot (due to slightly different field-of-view and the curvature of the object).

KarlAbson, 2018-12-03:
Hi @dorodnic 
Thanks for the link. That's really helpful!

closed 2018-12-03 by KarlAbson

1556260456, 2020-10-06:
I also have this problem. after I align to the depth frame, there are  a mount of black areas,just as follow picture,is there any way to avoid it?
/home/adam/Picture/heise.jpg