realsenseai/librealsense #2828: how can I get the extrinsics of particular picture?

anranx, 2018-12-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            | }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello, I want to ask some questions about calibration.
I downlowd the CalibrationToolAPI , and it returns the extrinsic between Right camera to Left camera and RGB camera to Left camera. I got the  RotationLeftRight TranslationLeftRight, RotationLeftRGB, TranslationLeftRGB.
What I want is the extrinsic of  pratical picture in left camera or in rgb camera. 
For example, I am taking  15 pictures of the checkerboard in infrared camera1 , I want to know the 15th picture's extrinsic. The Calibration Userbook said that the left camera is located in world origin, so I can get the 15th checkerboard_left camera Transformation matrix, including RotationLeft_15thPicture, TranslationLeft_15thPicture.
Or I  take 15 pictures of the checkerboard in rgb camera , so I can get the  15th  checkerboard_rgb camera Transformation matrix, including RotationRGB_15thPicture, TranslationRGB_15thPicture.
**Anyway, how can I get the extrinsics of particular picture instead of camera sensor?**

I try to put the RGB pictures in matlab calibrator, it returns  3*3*15 ratation and 3*15 translation, so I can get the 15th picture's extrinsic. But the intrinsic calculated in matlab is deferent in DyCalibRusult. For example, the FocalLenthRGB is 1392.89, 1393.84. But the  FocalLenthRGB in matlab is 1384. And the DistortionRGB=0, but  matlab has the Radial distortion. Also the PrincipalPointRGB is different.  
[image]
Due to the wrong instrinsic, so i think extrinsic is also wrong.
When i put the infrared pictures in matlab stereo calibrator, it refuses me a lot pictures（input 22 pairs remain 2 pairs pictures）and it shows the wrong cross point in checkerboard.
[image]
Due to the high rejection rate,  It's very difficult to get extrinsic and instrinsic.

So that is my problem, if anyone can tell me how to  I get the extrinsics of pratical picture ? Thank you!

anranx, 2018-12-04:
I also descripe this question in intel community:
https://communities.intel.com/thread/146282

RealSense-Customer-Engineering, 2018-12-27:
[Realsense Customer Engineering Team Comment]
Hi @anranx,

You use below configuration for D435 custom calibration, right?
1280x800, Y16 format, 15 FPS for Left/Right
1280x1080 YUY2 format, 15 FPS for RGB

From your picture, the IR emitter is on. Suggest to disable the IR emitter and you can use the environment light for the calibration. And, make sure auto exposure is on.

By the way, I temporarily have the access problem of the intel community link, so far, not able to see if any further info there.

closed 2019-01-07 by anranx