realsenseai/librealsense #2833: Mapping Depth to RGB

philipjames44, 2018-12-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |SR300 | 
| Operating System & Version |  MacOS  | 
| Platform                                 | PC |
| Language                            | C++                       | 
### Issue Description
Currently I identify objects from movement detected in the depth stream. I have previously done some calculations on focal length mappings to translate my points to the RGB image, however they are not very accurate. Is there any way to map the points from 2D depth to 2D RGB, or perhaps find the correct focal lengths? I calculated a focal length of 1.471 for x, and 1.333 for y.

matkatz, 2018-12-09:
Hi, 

Please check [rs-align Sample](https://github.com/IntelRealSense/librealsense/tree/master/examples/align) which demonstrate how to map between the depth and color frames.

philipjames44, 2018-12-09:
@matkatz 
Hi, thanks for responding. I previously looked at the rs-align sample, however I already have approximately 40 hours of footage stored in .bag files. So I'm trying to map from recorded data, whereas the rs-align sample seems to be only for live footage.

matkatz, 2018-12-10:
You can use the align filter on your recorded data, the filter also take care of the scaling between the streams.
The code below reads a bag file and map the color frames to the depth frames.
The processed frameset (named "data") includes the original color frame and an aligned depth frame with same resolution as the color frame.
```
#include <librealsense2/rs.hpp>
#include "example.hpp"

int main(int argc, char * argv[])
{
    // Create a simple OpenGL window for rendering:
    window app(1280, 720, "RealSense Capture Example");

    // Declare depth colorizer for pretty visualization of depth data
    rs2::colorizer color_map;
    rs2::align aligner(RS2_STREAM_COLOR); // map depth to color

    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_device_from_file("path_to_bag_file");
    pipe.start(cfg);

    while (app)
    {
        rs2::frameset data = pipe.wait_for_frames().
                             apply_filter(aligner).     // align depth to color   
                             apply_filter(color_map);   // Find and colorize the depth data
        app.show(data);
    }

    return EXIT_SUCCESS;
}
```

philipjames44, 2018-12-10:
@matkatz Oh wow that's very cool! Thank you. I applied it to my program and noticed a few distortions of depth information though. Would you happen to know if there is a way of retrieving the constants that align calculates to map depth to color?

matkatz, 2018-12-11:
You can access the intrinsic and extrinsic data in the following way:
```
    rs2::pipeline_profile p = pipe.start(cfg);
    rs2_intrinsics ci = p.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>().get_intrinsics();
    rs2_intrinsics di = p.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>().get_intrinsics();
    rs2_extrinsics d2c = p.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>().get_extrinsics_to(p.get_stream(RS2_STREAM_COLOR));
```
**Or by:**
```
        rs2::frame color = data.first(RS2_STREAM_COLOR);
        rs2::frame depth = data.first(RS2_STREAM_DEPTH, RS2_FORMAT_Z16);

        rs2_intrinsics ci = color.get_profile().as<rs2::video_stream_profile>().get_intrinsics();
        rs2_intrinsics di = depth.get_profile().as<rs2::video_stream_profile>().get_intrinsics();
        rs2_extrinsics d2c = depth.get_profile().as<rs2::video_stream_profile>().get_extrinsics_to(color.get_profile());
```

philipjames44, 2018-12-11:
@matkatz 
Very helpful, thank you. I will try this next week.

closed 2018-12-11 by philipjames44