realsenseai/librealsense #2834: D435 segfault running samples

orgicus, 2018-12-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version |   MacOS (10.11.6) | 
| Platform                                 | Macbook Pro |
| SDK Version                            |  2.17.0                          | 
| Language                            |  {C/C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description

I've compiled `librealsense` from source and tried running the samples with a D435

```bash
./realsense-viewer/realsense-viewer 
 03/12 17:46:41,645 WARNING [0x7000003a6000] (sensor.cpp:338) Unregistered Media formats : [ P UYVY ]; Supported: [ ]
Segmentation fault: 11
```

I've tried other samples that try to stream frames and they all segfault as well.

The `rs-enumerate-devices ` sample works just fine.

How can I use the D435 on OSX ?

Thank you,
George

RealSense-Customer-Engineering, 2018-12-08:
[Realsense Customer Engineering Team Comment]
Hi orgicus,
The RealSense-viewer should work on OSX.
One thing I noticed is that MacOS (10.11.6) does not fall under the supported platform.
The SDK supports Mac OS (High Sierra 10.13.2).

orgicus, 2018-12-08:
HI @RealSense-Customer-Engineering ,

Thanks for getting back.
I will try again Monday on a separate OSX machine which has already 10.3 installed.

If that works as expected I can plan for an OSX update, otherwise not looking forward to fixing
potential library + depedency issues with OpenCV 4, PCL, OpenPose, etc. all compiled from source with CUDA support.

closed 2019-01-14 by RealSense-Customer-Engineering