realsenseai/librealsense #2835: C++ Rosbag: Failed to create ros reader: Bag unindexed

HippoEug, 2018-12-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version | Win 10 | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            | 2.16.5 | 
| Language                            | C++  | 
| Segment			| others                   | 

# Introduction
Hello, I am trying to do a short `rosbag` recording.

# Goal
When a recording is done, with `record.bag`, I want to use `rs-convert.exe` to produce `realsense_Color_7.png`, `realsense_Depth_5.png`, and `realsense_Depth_5.csv`.

# Issue
I am unable to record and open `.bag` files **consistently**. Without modifying my short `rosbag` recording code, I can sometimes open the `.bag`, but other time it gives me an error of
`Failed to load file C:\Users\efoo1\Documents...\`
`Failed to create ros reader: Bag unindexed`

[image]

#### ROSBAG Code:
```
if (camera_button_rosbag) {
	pipe->start(recordConfig);
	std::this_thread::sleep_for(std::chrono::milliseconds(1500));
	pipe->stop();

	std::cout << "Done Recording (ROSBAG)" << std::endl;
}
```

[image]

The pipeline is stopped successfully (or so I think), why would I still have the bag unindexed error? Sometimes I wouldn't have this error, sometimes I would. I am unable to find out what is causing this, since this is random and isn't consistent..

Any help is appreciated, thanks!

dorodnic, 2018-12-04:
Hi @HippoEug 
Yes, this is definitely destruction issue, since ROS-bag indexing is performed when closing the file.
Can you try `pipe->stop(); pipe.reset();` to make sure everything is cleaned-up?
This might be a bug, if the recorder remains open after pipeline gets destroyed, so please let us know.

dorodnic, 2018-12-11:
Hi @HippoEug 
After looking into it, it seems that `config` object also holds the device open.
```cpp
{ // Note the added scope
     rs2::config convertConfig;
     convertToCSV.push_back(std::make_shared<rs2::tools::converter::converter_csv>("abc"));
     convertToPNG.push_back(std::make_shared<rs2::tools::converter::converter_png>("abc"));
     convertConfig.enable_device_from_file("record.bag");

     pipe->start(convertConfig);
}
```
I think this might be related.

HippoEug, 2018-12-13:
Hi @dorodnic 

I am still pretty confused. Firstly, what changes would the `{ }` do? Secondly, since I am reading from the file, isn't it appropriate to use `convertConfig.enable_device_from_file();` instead of `convertConfig.enable_record_to_file();`?

I did some modifications:
```
{ // Note the added scope
     rs2::config convertConfig;
     convertToCSV.push_back(std::make_shared<rs2::tools::converter::converter_csv>("abc"));
     convertToPNG.push_back(std::make_shared<rs2::tools::converter::converter_png>("abc"));
     convertConfig.enable_record_to_file("record.bag");
     pipe->start(convertConfig);
}

auto device = pipe->get_active_profile().get_device();
rs2::playback playback = device.as<rs2::playback>();
playback.set_real_time(false);
auto duration = playback.get_duration();
int progress = 0;
int frameNumber = 0ULL;

while (true) {
	std::cout << "Waiting for frames" << std::endl;
	auto frameset_c = pipe->wait_for_frames();
	std::cout << frameset_c[0].get_frame_number() << std::endl;

	if (frameset_c[0].get_frame_number() > 5) { // If end of frame, bag loops again, hence get_frame_number resets to 1 and is smaller
		std::for_each(convertToCSV.begin(), convertToCSV.end(), [&frameset_c](std::shared_ptr<rs2::tools::converter::converter_base>& converter) {
			converter->convert(frameset_c);
		});
		std::cout << "\nCONVERTING CSV" << std::endl;
		std::for_each(convertToCSV.begin(), convertToCSV.end(), [](std::shared_ptr<rs2::tools::converter::converter_base>& converter) {
			converter->wait();
		});
		std::cout << "CONVERTING PICTURES" << std::endl;

...
...
...

}
```

Running this code segment right after doing a ROSBAG recording with pipeline successfully closed still causes this error..

armandok, 2018-12-25:
I am having the same issue. First, I tried to open a saved rosbag file, which is saved by a different application, and the poll_for_frames returns false and I cannot read anything from the file.
When I try to open the bag file after recording it (in the same application), I get the `Bag unindexed` error.

HippoEug, 2019-01-07:
@armandok Yeap, I got to close the application after recording, then re-open again to open the recording. Have you found a workaround?

armandok, 2019-01-09:
@HippoEug Unfortunately not.
I am trying to record a session, and then load it later and align and filter the recorded images, because doing the align and filtering altogether in real-time reduces the fps to almost half.
The weird thing is that I can open the rosbag file with Realsenseviewer but when I open it within my application the poll_for_frames returns false. I have no idea what to do.

HippoEug, 2019-01-09:
I can open the rosbag file with Realsense Viewer and my own application **IF** I close out my application immediately after recording the rosbag file (using my application). If I attempt to open the file after rosbag recording is done without restarting my application, it will crash on my application, and Realsense Viewer would give the bag unindexed error. 

What I have to do currently on my application:
1. Open application
2. Press ROSBAG record
3. Close and start application again
4. Process that particular ROSBAG file

Seems like there is no way to skip step 3 at the moment 

Hence I am still thinking there is something in the destructor that is causing this, just that I do not know what is.

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi HippoEug,

Wonder if there is any updates from you on this one?

Thanks!

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi HippoEug,

Any update from your side?

Thanks!

realsens01, 2019-04-30:
If I'm not mistaken, this issue has the same underlying cause as [our issue with the Python wrapper](https://github.com/IntelRealSense/librealsense/issues/3044#issuecomment-473011378) (i.e., the bag is not written and closed when the pipeline is stopped). That issue still exists for me, even though there are ways to work around it (see that thread).

RealSenseCustomerSupport, 2019-08-07:
----------------------------------------------

@realsens01 @HippoEug Bag unindexed is caused by recording being unexpectedly interrupted or not properly closed. Did you get the same issue using provided playback-record demo?

realsens01, 2019-08-07:
Which playback-record demo? From the thread that I linked, or this thread? As discussed in this thread, it seems like an object destruction issue. If I understand correctly, something is holding the bag open past the point at which the person doing the coding expects it to be, so the index is not written. Is there a reason that the bag is not indexed when ``pipeline.stop()`` is called?

RealSenseCustomerSupport, 2019-08-16:
----------------------------------------------

@realsense01 Please find record-playback demo in https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback

closed 2019-08-29 by RealSenseCustomerSupport

realsens01, 2019-08-29:
Not really sure why this was closed. The issue remains and my question was not answered. [dorodnic referred to this as an unresolved issue](https://github.com/IntelRealSense/librealsense/issues/3029#issuecomment-453114359), and I continue to have problems with it. Maybe because the OP is not responding? I'll go ahead and make a new one.