realsenseai/librealsense #2838: Multicamera Depth Camara Setup --- RealSence Four-Camera Demo

PengStrongLee, 2018-12-04:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 x 3 | 
| Firmware Version                   | 2.16.2 | 
| Operating System & Version |   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |2                         | 
| Language                            |  C++ }                          | 

 Issue Description
I found a doc here: https://realsense.intel.com/multiple-depth-cameras-setup-guide/
I want to know what should I do to get the result just like this and where to find the "RealSence Four-Camera Demo" sample code.
Thanks

MartyG-RealSense, 2018-12-04:
Hi @PengStrongLee - this topic was discussed last week on the RealSense forum at the Intel Support site.

https://communities.intel.com/thread/146016

closed 2018-12-31 by RealSense-Customer-Engineering

fburak, 2020-07-30:
> 
> 
> Hi @PengStrongLee - this topic was discussed last week on the RealSense forum at the Intel Support site.
> 
> https://communities.intel.com/thread/146016

Hello @MartyG-RealSense , the link seems dead, could you provide new one?

MartyG-RealSense, 2020-07-30:
Hi @fburak  I cannot find a working link to that precise case but I know what I would have said.  The code of the 4-camera demo in the multiple cameras paper was not available to the public.  I believe it was created just for the purposes of illustrating multi-camera use in the paper.

You can instead set up multi-camera hardware sync in the RealSense Viewer program.  First, add each of your multiple cameras in the RealSense Viewer by clicking the **Add Source** button at the top of the Viewer's options side-panel, so that you end up with a list of cameras that each have their own set of controls.

[image]

Next, expand open the **Stereo Module** options category on each camera, and then expand open the **Controls** sub-category.

At the bottom of the Controls category, you will find an option called **Inter Cam Sync Mode**.  The option is set to '0' by default.

[image]

On one of the cameras, you should designate it as the **Master** camera that the other cameras will synchronize with by setting Inter Cam Sync Mode to '1' instead of '0'.

For all of the other cameras, you should set their Inter Cam Sync Mode to '2'.  This defines them as **Slave** cameras that will synchronize their timing with the Master camera.