realsenseai/librealsense #2840: D435 can't detect the depth on the phone?

laolaolulu, 2018-12-04:
[image]

laolaolulu, 2018-12-05:
Modifying the laser power value is still ineffective, as long as the plane reflectance can not measure the depth value, what is a good solution?

p0wdrdotcom, 2018-12-06:
Post processing may help you here.  Essentially filling the holes in the stereo data by guessing what it should be.  

I've experimented hole filling by segmenting the object, identifying the surface normal and then filling the data with the Z value that should fit the plane.

[image]
[image]

p0wdrdotcom, 2018-12-06:
You could also try lowering the exposure of the IR cameras to lower the effect of reflections.

RealSense-Customer-Engineering, 2018-12-13:
[Realsense Customer Engineering Team Comment]
@laolaolulu Can you try the post processing to fill the hole and see if it meets your need? Thanks!

RealSense-Customer-Engineering, 2018-12-20:
[Realsense Customer Engineering Team Comment]
@laolaolu Any update for this issue? Looking forward to your update. Thanks!

closed 2018-12-20 by laolaolulu