realsenseai/librealsense #2844: rs2_depth_frame_get_distance doesn't work for software_device sensor

ruanjiandong, 2018-12-04:
Repro Steps:
1. create software_device pipeline like https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device
2. add the following processing logic:
```
   rs2::align aligner(RS2_STREAM_COLOR);
   auto fset = sync.wait_for_frames();

   // the following succeeds since it use RS2_OPTION_DEPTH_UNITS
   aligner.process(fset);
   auto depth_frame = aligned_frame.get_depth_frame();

   // the following always return 0.0 since it doesn't use RS2_OPTION_DEPTH_UNITS
   auto depth = rs2_depth_frame_get_distance(depth_frame.get(), u, v, nullptr);
```
Root cause:
Internally, rs2_depth_frame_get_distance calls depth_frame::query_units function in src/archive.h. depth_frame::query_units can't correctly handle software depth sensor.

Proposed fix: depth_frame::query_units query RS2_OPTION_DEPTH_UNITS option on the sensor object itself.

dorodnic, 2018-12-05:
Hi @ruanjiandong 
Thank you for the report, we will look into this. The logic behind Depth-Units is a bit complex because we are concerned about sending hardware command to the device on every frame (it would be very bad in terms of both latency and power-consumption). So we try to cache it. We will look into how to solve this.

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

Please check librealsense 2.19.0 with the fix.

closed 2019-03-14 by RealSenseCustomerSupport