realsenseai/librealsense #2845: System.Exception:“Frame didn't arrived within 5000 rs2_pipeline_wait_for_frames(pipe:0000024B6EC2A350)”

laolaolulu, 2018-12-05:
Use alignment to run for a while (about 10 seconds) and then report an error. It works without alignment.

align = new Align(Intel.RealSense.Stream.Color);
var processed = align.Process(frames);

[image]

HippoEug, 2018-12-05:
What firmware are you using, have you tried updating the firmware or SDK version? Is the camera detected as USB 3, and are you using the original cable whatsoever? 👍

laolaolulu, 2018-12-05:
The latest firmware, the latest sdk, USB 3.2 
[image]

dorodnic, 2018-12-05:
Hi @laolaolulu 
Please note that when using the .NET wrapper, any object derived from `Frame` has to be either:
* Wrapped in a `using(...){ }` block
* Properly released by calling `.Dispose()`

This is most likely the root cause, with too many frames still being in use by .NET (waiting to be garbage-collected)