realsenseai/librealsense #2847: realsense-viewer: failed to view depthcloud (texture source: color)

sharronliu, 2018-12-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.14 | 
| Operating System & Version |   Linux (Ubuntu 16  | 
| Kernel Version (Linux Only)    | 4.8                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.14 }                          | 
| Language                            |      c/c++                      | 
| Segment			|  Robot                         | 

### Issue Description
In realsense-viewer, failed to display 3D depthcloud (with color selected as texture source). In the same environment, some other D400 cameras working. Does it imply hardware or firmware issue?
[image]
[image]

dorodnic, 2018-12-05:
Hi @sharronliu 
Could you please share the output of `rs-enumerate-devices -c` (for calibration)?

sharronliu, 2018-12-06:
@dorodnic attached the outputs.
[calibration_results.zip](https://github.com/IntelRealSense/librealsense/files/2651528/calibration_results.zip)

RealSense-Customer-Engineering, 2019-01-15:
[Realsense Customer Engineering Team Comment]
Hi sharronliu,

The output shows the color camera does not contain proper calibration values.
Could you please try to run the latest Dynamic Calibration tool to restore the calibration table?
https://realsense.intel.com/intel-realsense-downloads/#software

RealSenseCustomerSupport, 2019-01-31:
----------------------------------------------

Hi sharronliu,

Do you have chance to try the Dynamic Calibration yet?

RealSenseCustomerSupport, 2019-02-14:
----------------------------------------------

Hi sharronliu,
The RealSense Depth Quality Tool can be used to test the camera's depth quality, The tool and detail description of the metrics can be found in:
https://github.com/IntelRealSense/librealsense/tree/master/tools/depth-quality

ahuizxc, 2019-02-15:
After re-calibration, this problem was solved, thanks :).

closed 2019-02-23 by RealSenseCustomerSupport

mkhansen-intel, 2019-03-11:
I'm not finding a Linux version of the Dynamic Calibrator. Is there one somewhere?

ahuizxc, 2019-03-12:
> I'm not finding a Linux version of the Dynamic Calibrator. Is there one somewhere?

Hi matt

Here is the link you need

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400_Dyn_Calib_User_Guide.pdf

shenghui