realsenseai/librealsense #2850: xioctl(UVCIOC_CTRL_QUERY) Device or resource busy errors with kernel 4.19.5

sh0, 2018-12-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430 | 
| Firmware Version                   | 5.10.13.0 | 
| Operating System & Version |   Linux (Gentoo)  | 
| Kernel Version (Linux Only)    | 4.19.5 | 
| Platform                                 | PC  |
| SDK Version                            |  master branch at 79dda7f | 
| Language                            |  C++ | 
| Segment			|  N/A | 

### Issue Description
On Linux kernel 4.19.5 errors are generated when some commands are sent to camera in too quick succession. The error message is: `ERROR [139644039424128] (types.h:188) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy`

Most reliable way to trigger this error is to create an `rs2::pipeline`, start it with `start()` and immediately use `stream1.get_extrinsics_to(stream2)` with appropriate streams. The `get_extrinsics_to` will throw with above error.

Current workaround is to wait around 3 seconds after starting the pipeline so that the camera can settle and then run `get_extrinsics_to` without issues.

thestr4ng3r, 2018-12-08:
I can confirm this on `4.19.4-arch1-1-ARCH`.

thestr4ng3r, 2018-12-08:
Here's a VERY VERY cheap and dirty workaround:
```
diff --git a/src/linux/backend-v4l2.cpp b/src/linux/backend-v4l2.cpp
index 3366845e..8eca827d 100644
--- a/src/linux/backend-v4l2.cpp
+++ b/src/linux/backend-v4l2.cpp
@@ -1058,8 +1058,11 @@ namespace librealsense
         {
             uvc_xu_control_query q = {static_cast<uint8_t>(xu.unit), control, UVC_GET_CUR,
                                       static_cast<uint16_t>(size), const_cast<uint8_t *>(data)};
+            retry:
             if(xioctl(_fd, UVCIOC_CTRL_QUERY, &q) < 0)
             {
+                if(errno == EBUSY)
+                    goto retry;
                 if (errno == EIO || errno == EAGAIN) // TODO: Log?
                     return false;
```

dorodnic, 2018-12-10:
Hi @sh0 @thestr4ng3r 
I tested the camera with 4.19 RC about a month ago. 
Thank you for the report, we will have to look into it.

dsharma2701, 2018-12-10:
Am getting the same error on D435 with all the tools : rs-capture, realsense-viewer etc 

uname -a
4.19.0-041900-generic #201810221809 SMP Mon Oct 22 22:11:45 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

Librealsense ->  2.17.0 and realsense2_camera -> 2.1.1 
Both of them were built from source

dsharma2701, 2018-12-10:
Is there a combination of the versions from sdk and driver which works with the 4.19 kernel ?

dorodnic, 2018-12-11:
At the moment we do not officially support it **yet**: [v2.17.0 - Supported Platforms](https://github.com/IntelRealSense/librealsense/releases/tag/v2.17.0)
Obviously, we will do our best to get on top of this ASAP. 
In the meantime, you have the option of running with one of the LTS kernels, trying @thestr4ng3r work-around or building with `-DFORCE_LIBUVC=true`.

gaborpapp, 2019-01-03:
encountered the same issue on Fedora 29. building with `-DFORCE_LIBUVC=true` resulted in a crash. the workaround works, although the system is continuously spitting out error messages while running `realsense-viewer`.

output of `dmesg`
```
[ 1169.365960] Unknown pixelformat 0x00000000
[ 1169.366016] WARNING: CPU: 5 PID: 9092 at drivers/media/v4l2-core/v4l2-ioctl.c:1340 v4l_enum_fmt+0xfc0/0x1390 [videodev]
[ 1169.366017] Modules linked in: fuse rfcomm ccm xt_CHECKSUM ipt_MASQUERADE tun bridge stp llc devlink nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 xt_conntrack ebtable_nat ip6table_nat nf_nat_ipv6 ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_nat_ipv4 nf_nat iptable_mangle iptable_raw iptable_security nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c ip_set nfnetlink ebtable_filter ebtables ip6table_filter ip6_tables bnep bbswitch(OE) sunrpc arc4 iwlmvm snd_hda_codec_hdmi mac80211 intel_rapl x86_pkg_temp_thermal intel_powerclamp iTCO_wdt hid_multitouch coretemp iTCO_vendor_support snd_hda_codec_realtek snd_hda_codec_generic kvm_intel snd_hda_intel intel_cstate snd_hda_codec intel_uncore iwlwifi snd_hda_core intel_rapl_perf btusb snd_hwdep
[ 1169.366094]  btrtl uvcvideo snd_seq btbcm btintel videobuf2_vmalloc videobuf2_memops snd_seq_device bluetooth videobuf2_v4l2 cfg80211 snd_pcm dcdbas videobuf2_common dell_wmi_descriptor wmi_bmof videodev joydev snd_timer thunderbolt media intel_wmi_thunderbolt snd soundcore i2c_i801 ecdh_generic idma64 rfkill mei_me intel_lpss_pci mei intel_pch_thermal intel_lpss pcc_cpufreq acpi_pad dm_crypt i915 kvmgt mdev vfio kvm irqbypass i2c_algo_bit drm_kms_helper mxm_wmi drm crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel r8169 i2c_hid wmi video pinctrl_cannonlake pinctrl_intel
[ 1169.366166] CPU: 5 PID: 9092 Comm: realsense-viewe Tainted: G        W  OE     4.19.13-300.fc29.x86_64 #1
[ 1169.366168] Hardware name: Razer Blade/DANA_MB, BIOS 01.01 08/31/2018
[ 1169.366180] RIP: 0010:v4l_enum_fmt+0xfc0/0x1390 [videodev]
[ 1169.366184] Code: c6 7d 95 59 c0 e9 55 f1 ff ff 48 c7 c6 74 98 59 c0 3d 50 57 43 32 0f 84 43 f1 ff ff 89 c6 48 c7 c7 29 99 59 c0 e8 ba f0 b2 c4 <0f> 0b 41 80 7c 24 0c 00 0f 85 45 f1 ff ff 41 8b 44 24 2c 49 8d 7c
[ 1169.366186] RSP: 0018:ffffaf15078dbcc8 EFLAGS: 00010282
[ 1169.366190] RAX: 0000000000000000 RBX: ffff9a29a7d2ce00 RCX: 0000000000000006
[ 1169.366193] RDX: 0000000000000007 RSI: 0000000000000086 RDI: ffff9a29ad956860
[ 1169.366195] RBP: 0000000000000000 R08: 0000000000000030 R09: 0000000000000003
[ 1169.366197] R10: 0000000000000000 R11: 0000000000000001 R12: ffffaf15078dbdc0
[ 1169.366203] R13: ffffffffc07ad9c0 R14: ffff9a280d495f00 R15: ffff9a29a7d2ce00
[ 1169.366206] FS:  00007fb1a4721ac0(0000) GS:ffff9a29ad940000(0000) knlGS:0000000000000000
[ 1169.366209] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1169.366212] CR2: 00007ff904212078 CR3: 00000004a7210005 CR4: 00000000003606e0
[ 1169.366214] Call Trace:
[ 1169.366229]  __video_do_ioctl+0x135/0x2f0 [videodev]
[ 1169.366236]  ? path_openat+0x3d0/0x1610
[ 1169.366242]  ? syscall_return_via_sysret+0x13/0x83
[ 1169.366255]  video_usercopy+0x253/0x5a0 [videodev]
[ 1169.366267]  ? copy_overflow+0x20/0x20 [videodev]
[ 1169.366275]  ? selinux_file_ioctl+0x161/0x200
[ 1169.366287]  v4l2_ioctl+0x45/0x50 [videodev]
[ 1169.366298]  do_vfs_ioctl+0xa4/0x630
[ 1169.366305]  ksys_ioctl+0x60/0x90
[ 1169.366311]  __x64_sys_ioctl+0x16/0x20
[ 1169.366317]  do_syscall_64+0x5b/0x160
[ 1169.366322]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 1169.366325] RIP: 0033:0x7fb1a558609b
[ 1169.366329] Code: 0f 1e fa 48 8b 05 ed bd 0c 00 64 c7 00 26 00 00 00 48 c7 c0 ff ff ff ff c3 66 0f 1f 44 00 00 f3 0f 1e fa b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d bd bd 0c 00 f7 d8 64 89 01 48
[ 1169.366331] RSP: 002b:00007fffecde6b38 EFLAGS: 00000202 ORIG_RAX: 0000000000000010
[ 1169.366335] RAX: ffffffffffffffda RBX: 00007fb190001680 RCX: 00007fb1a558609b
[ 1169.366337] RDX: 00007fffecde6c00 RSI: 00000000c0405602 RDI: 0000000000000012
[ 1169.366340] RBP: 00007fffecde7220 R08: 0000000000000004 R09: 0000000000000005
[ 1169.366342] R10: 2ce33e6c02ce33e7 R11: 0000000000000202 R12: 00007fb1a7439910
[ 1169.366344] R13: 00007fffecde97c0 R14: 0000000000000000 R15: 0000000000000000
```
`uname -a`
`Linux xxxrz 4.19.13-300.fc29.x86_64 #1 SMP Sat Dec 29 22:54:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`

d435 with 05.10.06.00 firmware and latest development branch

codeyash, 2019-01-16:
@thestr4ng3r Your fix worked! Thanks

FeiyingLan, 2019-01-23:
> > Here's a VERY VERY cheap and dirty workaround:
> > ```
> > diff --git a/src/linux/backend-v4l2.cpp b/src/linux/backend-v4l2.cpp
> > index 3366845e..8eca827d 100644
> > --- a/src/linux/backend-v4l2.cpp
> > +++ b/src/linux/backend-v4l2.cpp
> > @@ -1058,8 +1058,11 @@ namespace librealsense
> >          {
> >              uvc_xu_control_query q = {static_cast<uint8_t>(xu.unit), control, UVC_GET_CUR,
> >                                        static_cast<uint16_t>(size), const_cast<uint8_t *>(data)};
> > +            retry:
> >              if(xioctl(_fd, UVCIOC_CTRL_QUERY, &q) < 0)
> >              {
> > +                if(errno == EBUSY)
> > +                    goto retry;
> >                  if (errno == EIO || errno == EAGAIN) // TODO: Log?
> >                      return false;
> > ```
> 
> Hi, Can I ask how I can use this code?
> My realsense cannot work on my archlinux.
> I updated the firmware of realsense. But it shows the device is busy.
> Thanks!
> [image]

It works now.
I just unplug my realsense and plug it in another USB3.0 port. And it works!

vcarpani, 2019-03-28:
Any updates on this issue?

maxime-tournier, 2019-04-05:
@thestr4ng3r thanks for your workaround. I also had to apply a similar hack on function `set_xu`, otherwise it would only work on the first try after the camera is connected.

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

This issue is fixed in librealsense 2.24.0.

closed 2019-07-02 by RealSenseCustomerSupport

yyxr75, 2019-08-21:
Hi there,
I just encountered this issue yesterday.
My system is ubuntu-18.04
`uname -a`
`Linux yang-B250-HD3P 5.0.0-25-generic #26~18.04.1-Ubuntu SMP Thu Aug 1 13:51:02 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

And librealsense version is the latest one showed below:

[image]

and the error is:
`terminate called after throwing an instance of 'rs2::backend_error'
  what():  set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
Aborted (core dumped)`

Any suggestion will be appreciated!!

By the way, I dont know how to implement his method above
https://github.com/IntelRealSense/librealsense/issues/2850#issuecomment-445460696
would please tell me how!

maximilianwulf, 2019-08-21:
@yyxr75, I got the same error on my laptop and the reason was that Ubuntu updated the kernel to 5.0. ... without me noticing it. Switching back to 4.18 solved the problem.
Supported kernel versions are listed here https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md.

yyxr75, 2019-08-23:
> @yyxr75, I got the same error on my laptop and the reason was that Ubuntu updated the kernel to 5.0. ... without me noticing it. Switching back to 4.18 solved the problem.
> Supported kernel versions are listed here https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md.

Thanks a lot. So much help!