realsenseai/librealsense #2855: multi time run intel-ros/realsense, uvcvideo: Non-zero status (-71) happens

FENGZHANG123, 2018-12-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415/ D435} | 
| Firmware Version                   | (5.10.6) | 
| Operating System & Version |   { Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    |  (4.15.0-42-generic)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<16>.<3> }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot}                          | 
| intel-ros/realsense        |{2.11.1}|

### Issue Description
Hi, when I run realsense2_camera first time, everything is fine.But when I shutdown the program with 'Ctrl C' and restart it, error happens:
PARAMETERS
 * /downwire_camera/realsense2_camera/accel_fps: 1000
 * /downwire_camera/realsense2_camera/accel_optical_frame_id: downwire_camera_a...
 * /downwire_camera/realsense2_camera/align_depth: False
 * /downwire_camera/realsense2_camera/aligned_depth_to_color_frame_id: downwire_camera_a...
 * /downwire_camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: downwire_camera_a...
 * /downwire_camera/realsense2_camera/aligned_depth_to_infra1_frame_id: downwire_camera_a...
 * /downwire_camera/realsense2_camera/aligned_depth_to_infra2_frame_id: downwire_camera_a...
 * /downwire_camera/realsense2_camera/base_frame_id: downwire_camera_link
 * /downwire_camera/realsense2_camera/color_fps: 30
 * /downwire_camera/realsense2_camera/color_frame_id: downwire_camera_c...
 * /downwire_camera/realsense2_camera/color_height: 480
 * /downwire_camera/realsense2_camera/color_optical_frame_id: downwire_camera_c...
 * /downwire_camera/realsense2_camera/color_width: 848
 * /downwire_camera/realsense2_camera/depth_fps: 30
 * /downwire_camera/realsense2_camera/depth_frame_id: downwire_camera_d...
 * /downwire_camera/realsense2_camera/depth_height: 480
 * /downwire_camera/realsense2_camera/depth_optical_frame_id: downwire_camera_d...
 * /downwire_camera/realsense2_camera/depth_width: 848
 * /downwire_camera/realsense2_camera/enable_color: True
 * /downwire_camera/realsense2_camera/enable_depth: True
 * /downwire_camera/realsense2_camera/enable_fisheye: True
 * /downwire_camera/realsense2_camera/enable_imu: True
 * /downwire_camera/realsense2_camera/enable_infra1: True
 * /downwire_camera/realsense2_camera/enable_infra2: True
 * /downwire_camera/realsense2_camera/enable_pointcloud: False
 * /downwire_camera/realsense2_camera/enable_sync: False
 * /downwire_camera/realsense2_camera/filters: 
 * /downwire_camera/realsense2_camera/fisheye_fps: 30
 * /downwire_camera/realsense2_camera/fisheye_height: 480
 * /downwire_camera/realsense2_camera/fisheye_optical_frame_id: downwire_camera_f...
 * /downwire_camera/realsense2_camera/fisheye_width: 848
 * /downwire_camera/realsense2_camera/gyro_fps: 1000
 * /downwire_camera/realsense2_camera/gyro_optical_frame_id: downwire_camera_g...
 * /downwire_camera/realsense2_camera/infra1_fps: 30
 * /downwire_camera/realsense2_camera/infra1_frame_id: downwire_camera_i...
 * /downwire_camera/realsense2_camera/infra1_height: 480
 * /downwire_camera/realsense2_camera/infra1_optical_frame_id: downwire_camera_i...
 * /downwire_camera/realsense2_camera/infra1_width: 848
 * /downwire_camera/realsense2_camera/infra2_fps: 30
 * /downwire_camera/realsense2_camera/infra2_frame_id: downwire_camera_i...
 * /downwire_camera/realsense2_camera/infra2_height: 480
 * /downwire_camera/realsense2_camera/infra2_optical_frame_id: downwire_camera_i...
 * /downwire_camera/realsense2_camera/infra2_width: 848
 * /downwire_camera/realsense2_camera/json_file_path: 
 * /downwire_camera/realsense2_camera/pointcloud_texture_index: 0
 * /downwire_camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /downwire_camera/realsense2_camera/rosbag_filename: 
 * /downwire_camera/realsense2_camera/serial_no: 827312071232
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /downwire_camera/
[... 123 lines cut ...]
 06/12 17:29:39,879 WARNING [139908659517312] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1544088579.880634647]: depth stream is enabled - width: 848, height: 480, fps: 30
[ INFO] [1544088579.880881410]: infra1 stream is enabled - width: 848, height: 480, fps: 30
[ INFO] [1544088579.881104228]: infra2 stream is enabled - width: 848, height: 480, fps: 30
 06/12 17:29:39,881 WARNING [139908659517312] (backend-v4l2.cpp:1248) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 06/12 17:29:39,881 WARNING [139908659517312] (sensor.cpp:338) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1544088579.883141575]: color stream is enabled - width: 848, height: 480, fps: 30
 06/12 17:29:44,891 WARNING [139907938899712] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
 06/12 17:29:44,975 ERROR [139908659517312] (types.h:179) xioctl(VIDIOC_STREAMON) failed for buf_type=1 Last Error: Input/output error
[ERROR] [1544088590.097255257]: An exception has been thrown: xioctl(VIDIOC_STREAMON) failed for buf_type=1 Last Error: Input/output error
[ERROR] [1544088590.097558169]: An exception has been thrown: xioctl(VIDIOC_STREAMON) failed for buf_type=1 Last Error: Input/output error
[FATAL] [1544088590.245427995]: Failed to load nodelet '/downwire_camera/points_xyzrgb_sw_registered` of type `depth_image_proc/point_cloud_xyzrgb` to manager `realsense2_camera_manager'
[FATAL] [1544088590.245463713]: Failed to load nodelet '/downwire_camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[FATAL] [1544088590.245472823]: Failed to load nodelet '/downwire_camera/color_rectify_color` of type `image_proc/rectify` to manager `realsense2_camera_manager'
[FATAL] [1544088590.245516913]: Failed to load nodelet '/downwire_camera/register_depth` of type `depth_image_proc/register` to manager `realsense2_camera_manager'
[downwire_camera/realsense2_camera_manager-2] process has died [pid 16466, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-realsense2_camera_manager-2.log].
log file: /home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-realsense2_camera_manager-2*.log
[downwire_camera/realsense2_camera-3] process has died [pid 16467, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-realsense2_camera-3.log].
log file: /home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-realsense2_camera-3*.log
[downwire_camera/color_rectify_color-4] process has died [pid 16468, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load image_proc/rectify realsense2_camera_manager --no-bond image_mono:=color/image_raw image_rect:=color/image_rect_color __name:=color_rectify_color __log:=/home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-color_rectify_color-4.log].
log file: /home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-color_rectify_color-4*.log
[downwire_camera/register_depth-5] process has died [pid 16476, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/register realsense2_camera_manager false rgb/camera_info:=color/camera_info depth/camera_info:=depth/camera_info depth/image_rect:=depth/image_rect_raw depth_registered/image_rect:=depth_registered/sw_registered/image_rect_raw __name:=register_depth __log:=/home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-register_depth-5.log].
log file: /home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-register_depth-5*.log
[downwire_camera/points_xyzrgb_sw_registered-6] process has died [pid 16485, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb realsense2_camera_manager false rgb/image_rect_color:=color/image_rect_color rgb/camera_info:=color/camera_info depth_registered/image_rect:=depth_registered/sw_registered/image_rect_raw depth_registered/points:=depth_registered/points __name:=points_xyzrgb_sw_registered __log:=/home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-points_xyzrgb_sw_registered-6.log].
log file: /home/zf/.ros/log/73d9914a-f939-11e8-a579-58fb84344185/downwire_camera-points_xyzrgb_sw_registered-6*.log

RealSense-Customer-Engineering, 2018-12-08:
[Realsense Customer Engineering Team Comment]
Hello @FENGZHANG123

A few things:

-  The other github issue you linked: https://github.com/IntelRealSense/librealsense/issues/1615 was solved by ensuring enough power was being delivered to the RealSense camera.  The customer was using a faulty USB hub...

-  Please check the following:

_  >  Have you tried different USB 3.0 type-C cables and plugging those in different ports?_

_  >  If you are using a USB 3.0 hub to connect RealSense cameras, ensure hub has an external power source._

-  Also, if the issue you are seeing can be fixed by unplugging and re plugging the camera, try resetting the device by:

  >  Resetting device with C++ RealSense API:
https://github.com/IntelRealSense/librealsense/issues/1086

Code snippet:

rs2::device dev = device_you_are_using(); dev.hardware_reset();

//Will disconnect the camera and reconnect it. //NOTE: All camera settings will be restored to default after this call (same as unplug->plug)

//Don't use device after this line

   >  Resetting device with Python:

Code snippet:

import usb.core

from usb.core import find as finddev

dev = finddev(idVendor=0x8086, idProduct=0x0b07) dev.reset()

#Use lsusb to find the hardware IDs of Realsense camera, use those for the IDs in this reset script.

#D435 camera hardware IDs used for this script.

Please let us know if you have any other issues.

Thank you!

FENGZHANG123, 2018-12-10:
Sorry for delayed response, thank you for point 1-2-3. 

I used desktop PC, this problem never happened,  but on low-powered industrial control PC we used actually, the situation was just I said before. I checked industrial PC's voltage with multimeter, it was fine. Maybe the power wasn't stable. I haven't got any mind to measure the power for a long time.

I tried image camera's USB 3.0 type-C, nothing improved.

I suspected it was uvcvideo problem also one time cause 'dmesg' told me uvcvideo error and then realsense D435/D415 disconnected, that's why I had to replug the depth camera.

Point 3 seemed can solve my project problem for now,  I'll try to figure out what is the exact problem happened here later. Thank you!

FENGZHANG123, 2018-12-11:
@RealSense-Customer-Engineering Thank you,  those codes did work.

closed 2018-12-11 by FENGZHANG123