realsenseai/librealsense #2858: Is there any way to set distance limit for capturing point cloud ? (c++)

chaiban, 2018-12-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

FENGZHANG123, 2018-12-07:
PCL(Point Cloud Library) has a API named '[PassThrough Filter](http://docs.pointclouds.org/trunk/classpcl_1_1_pass_through.html)' can satisfy your need.

chaiban, 2018-12-07:
Thank you so much

closed 2018-12-07 by chaiban

jb455, 2018-12-07:
An in-SDK way to do it would be to set the Depth Clamp max/min in advanced mode>depth table. The only example I can find for advanced mode is in Python (https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-rs400-advanced-mode-example.py) but that should be fairly easy to translate to different languages.

chaiban, 2018-12-07:
Thank you so much but it works beautifully with PassThrough FIlter. I tested it