realsenseai/librealsense #2860: pointcloud processing block causing memory leak

mf-bwf, 2018-12-07:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    | 4.15.0-42-generic | 
| Platform                                 | PC |
| SDK Version                            |  v2.16.5         | 
| Language                            |  c++                          | 
| Segment			|  any | 

### Issue Description
There seems to be an issue with memory deallocation for rs2::pointcloud
The following minimum example code results in a memory leak.

The memory leak can be made smaller by moving the definition of 'pc' outside of the while loop, keeping the instance alive until function exit.

```cpp
void minimumTest()
{    
    rs2::pipeline pipe;
    cv::Mat pc_mat;

    // start stream
    rs2::pipeline_profile profile = pipe.start();

    // run for a while
    int i = 0;
    while(i < 10000)
    {
        // get a set of frames
        rs2::frameset update = pipe.wait_for_frames();
        rs2::frame depth;

        if(depth)
        {
            // create pointcloud
            rs2::pointcloud pc;
            rs2::points points = pc.calculate(depth);
            cv::Mat(720, 1280, CV_32FC3, (void*)points.get_data()).copyTo(pc_mat);

            // display pointcloud
            cv::imshow("minimum", pc_mat);
            if(cv::waitKey(1) > 0)
            {
                std::cout << "bye" << std::endl;
                break;
            }
        }
        i++;
    }
}
```

dorodnic, 2018-12-09:
Hi @mf-bwf 
Thanks for the report. 
The way it works, every instance of `pointcloud` is generating a stream-profile that gets registered within the global extrinsics graph. The profile itself gets deleted (via a smart-pointer), but the instance of `weak_ptr` to it in the extrinsics graph remains until an explicit clean-up. These happen whenever you re-connect the device. We could trigger these also for every creation / destruction of a processing block to solve this issue.
However, creating the `pointcloud` object within a loop is not recommended (for anything except reproducing the leak). In most cases, you will be Ok with just one (or one per device).

denkywu, 2019-04-26:
I have the same problem, and also solve the problem using the same way.
SDK Version: 2.19.0.570

OneQuestion, 2020-01-28:
So, exactly, what is the proper way of avoiding any such memory leak? What is the procedure of destroying processing block?

kmateti, 2021-03-22:
> Hi @mf-bwf
> Thanks for the report.
> The way it works, every instance of `pointcloud` is generating a stream-profile that gets registered within the global extrinsics graph. The profile itself gets [...]

Hi, 

I was wondering if there was any update on this issue?  I have found the memory leak even without any `pointcloud` instance, shown here (running v2.33.1):

```cpp
#include <librealsense2/rs.hpp>
#include <iostream>

int main()
{
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_device_from_file(<insert bag file path here>);
    pipe.start(cfg);
    auto frames = pipe.wait_for_frames();
    auto depth = frames.get_depth_frame();
    long int i = 0;
    while (1)
    {
        i++;
        std::cout << i << std::endl;
        frames = pipe.wait_for_frames();
        depth = frames.get_depth_frame();
    }
}
```

ev-mp, 2021-08-04:
Fix is in the `development` branch via #9553

closed 2021-08-04 by ev-mp