realsenseai/librealsense #2866: D435 reconnection issues

jasoncatchpole, 2018-12-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Windows 10  | 
| Kernel Version (Linux Only)    |  NA   | 
| Platform                                 | PC - Intel Core i7-4770, Nvidia 2070 RTX  |
| SDK Version                            |  2.16.3                          | 
| Language                            | C++            | 
| Segment			|  Kiosk application                        | 

### Issue Description
Hi there,

We have been building a kiosk application for a large customer and I have recently added functionality to our application to support the camera being disconnected and reconnected from the PC and having our application continue to work and start getting frames again when the camera is reconnected (or if the camera disappears from the USB bus and comes back again for any reason, we need this application to be robust as it will be operating for days). This reconnection handling is performed by doing a new "pipe.start(cfg)" using a configuration specifying both color and depth (at 640x480@30).

In this process, however, I have encountered a potential camera bug. After the camera is reconnected it starts receiving frames, and then all of a sudden (fairly often) the RGB frames being returned seem to 'get stuck', meaning that they start returning the same frame over and over (i.e. a still frame containing a snapshot of the scene), and yet the depth frames seem to continue fine. 

There doesn't seem to be anything I can do about this on the application code side, I am already checking the result from "wait_for_frames", it returns without exception when using a pipeline configured to return a color and depth stream. "processedFrameset.get_color_frame()" also returns a "rs2::frame" fine. The frame itself is valid its just stuck. 

I have seen this problem present after a few seconds once the camera is reconnected, I have also seen it where we don't get any new frames when the camera is plugged back it, i.e. it happens almost immediately. Even if I completely restart our application the problem persists, the camera itself seems to get into a bad state and needs to be power cycled to fix it.

Any thoughts on how to solve this (might need a firmware change) or get around it in the mean time? Happy to provide any further details to reproduce this or help solve it.

jasoncatchpole, 2018-12-11:
Hey @dorodnic, 

Thanks a lot for replying. I have reproduced the problem with debug logging on.

Here is [the log file](https://github.com/IntelRealSense/librealsense/files/2665806/realsense_logging_colour_def_stuck.log). I can't see much in there but it will probably mean a bit more to you.

dorodnic, 2018-12-12:
Hi @jasoncatchpole 

We went over the log and ruled-out some of the possibilities. 

One of the engineers asked to check if you are using non-standard RGB format (possibly YUY?). If that is the case, an upgrade to 2.17.0 might help.
However, assuming you are using RGB, from the log it looks like librealsense stops receiving RGB frames from the Media Foundation layer (without any visible error notification). 

There is also significant amount of OS errors related to device-tree enumeration. I have a suspicion this might be related. Can you share the code snippet you are using to detect disconnection and restart the stream? Is there an option to add a 30-sec delay between disconnection detection and start?

jasoncatchpole, 2018-12-14:
Hey @dorodnic,

You are correct that I am using RGB.

As you will be able to see below there is some delay between reconnection attempts. I could change this delay to anything we need. I could modify the code so that once the "IsRealsenseDeviceConnected()" function (in code below) returns true we wait 30 seconds before initialisation, its already waiting maybe 5-10 seconds roughly I would say.

Here are some code snippets for how I am initialising the camera, and how I detect when the camera has been disconnected and then do reinitialisation.

Thanks again.

``` 
// PIMPL
Impl()
{
    // initialise the camera
    isSuccessfullySetup = InitialiseCamera();
}

bool InitialiseCamera(bool suppressLogMessages = false)
{
    if (!IsRealsenseDeviceConnected())
    {
       // no realsense camera connected
        return false;
    }

    // Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;

    int resolutionToTryWidth = 640,
        resolutionToTryHeight = 480,
        framerateToTry = 30;

    bool setupStreamSuccess = false;
    bool haveTriedLowestResolution = false;

    if (isSuccessfullySetup)
    {
        // if this is the case it means we were previously processing frames and something happened
        pipe.stop();
        isSuccessfullySetup = false;
    }

    while (!setupStreamSuccess)
    {
        setupStreamSuccess = SetupConfiguration(cfg, resolutionToTryWidth, resolutionToTryHeight, framerateToTry, suppressLogMessages);
        if (!setupStreamSuccess)
        {
            if (!haveTriedLowestResolution)
            {
                if (!suppressLogMessages)
                    cout << "Failed to initialise camera with configured settings. Falling back to lowest supported resolution of 640x480 @ 15fps" << endl;
                resolutionToTryWidth = 640;
                resolutionToTryHeight = 480;
                framerateToTry = 15;
                haveTriedLowestResolution = true;
            }
            else
            {
                return false;
            }
        }
    }

    // Each depth camera might have different units for depth pixels, so we get it here
    // Using the pipeline's profile, we can retrieve the device that the pipeline uses
    float depth_scale = get_depth_scale(profile.get_device());
   

    rs2::stream_profile colourCameraStream = profile.get_stream(RS2_STREAM_COLOR);
[... 184 lines cut ...]
            continue;
        }

        // callback to provide the resulting images and use them
        OnNewRgbdFrame(color, depth);
    }

    // stop the pipeline
    try
    {
        pipe.stop();
    }
    catch (const rs2::error & e)
    {
        cout << "Caught realsense exception while attempting to stop the Realsense pipeline" << endl;
    }
    catch (const std::exception& e)
    {
        cout << "Caught exception while attempting to stop the Realsense pipeline" << endl;
    }

    isStopped = true;
    cout << "RealsenseSource stopping" << endl;
}
```

jasoncatchpole, 2019-01-16:
Hey @RealSense-Customer-Engineering,

Thanks for the response, sorry for my late response. I was waiting for an update from you so I hadn't tried the sleep yet, i'll try and test this out in the next couple of days.

Interesting that you couldn't reproduce, I managed to make this happen on a couple of the machines I tested here in our office.

I'll report back when I have added a larger sleep.

jasoncatchpole, 2019-01-16:
Incidentally, sometimes it takes many successive runs to reproduce this. I tried reproducing yesterday and disconnected/reconnected many times (perhaps as many as 10) and couldn't do it. Today I managed to do it second time. Just a note on how frequent (or infrequent) it is

RealSenseCustomerSupport, 2019-01-25:
Hi jasoncatchpole,

Thanks for the updates.
Glad to hear that the big sleep works for you.

Cheers!