realsenseai/librealsense #2868: rk3328-cc board with 18.04 install fail

dongxuanlb, 2018-12-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   |latest| 
| Operating System & Version |  Ubuntu 18.04(arch) | 
| Kernel Version (Linux Only)    |  4.4.114  | 
| Platform                                 | rk3328  |
| SDK Version                            |  with source }                          | 
| Language                            |  | 
| Segment			|  others }                          | 

### Issue Description
I'm trying to install sdk from source according [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md).

My platform is rk3288 and with the latest img:
1. ROC-RK3328-CC-ubuntu18.04-LXDE_20180605
2. Linux firefly 4.4.114 #46 SMP Tue Jun 5 17:30:08 HKT 2018 aarch64 aarch64 aarch64 GNU/Linux

And with the guide, I run the "script/patch-arch.sh", and and patching the file went wrong(realsense-camera-formats.patch).

I fixed it by :
`@@ -624,6 +624,18 @@ struct v4l2_pix_format {
 #define V4L2_PIX_FMT_Y8I      v4l2_fourcc('Y', '8', 'I', ' ') /* Greyscale 8-bit L/R interleaved */
 #define V4L2_PIX_FMT_Y12I     v4l2_fourcc('Y', '1', '2', 'I') /* Greyscale 12-bit L/R interleaved */
 #define V4L2_PIX_FMT_Z16      v4l2_fourcc('Z', '1', '6', ' ') /* Depth data 16-bit */
+#define V4L2_PIX_FMT_Y8       v4l2_fourcc('Y', '8', ' ', ' ') /* Greyscale 8-bit */
+#define V4L2_PIX_FMT_Y10      v4l2_fourcc('Y', '1', '0', ' ') /* Greyscale 10-bit */
+#define V4L2_PIX_FMT_Y12      v4l2_fourcc('Y', '1', '2', ' ') /* Greyscale 12-bit */
+#define V4L2_PIX_FMT_Y16      v4l2_fourcc('Y', '1', '6', ' ') /* Greyscale 16-bit */
+#define V4L2_PIX_FMT_RAW8     v4l2_fourcc('R', 'A', 'W', '8') /* Raw data 8-bit */
+#define V4L2_PIX_FMT_RW16     v4l2_fourcc('R', 'W', '1', '6') /* Raw data 16-bit */
+#define V4L2_PIX_FMT_INVZ     v4l2_fourcc('I', 'N', 'V', 'Z') /* 16 Depth */
+#define V4L2_PIX_FMT_INZI     v4l2_fourcc('I', 'N', 'Z', 'I') /* 24 Depth/IR 16:8 */
+#define V4L2_PIX_FMT_INVR     v4l2_fourcc('I', 'N', 'V', 'R') /* 16 Depth */
+#define V4L2_PIX_FMT_INRI     v4l2_fourcc('I', 'N', 'R', 'I') /* 24 Depth/IR 16:8 */
+#define V4L2_PIX_FMT_INVI     v4l2_fourcc('I', 'N', 'V', 'I') /* 8 IR */
+#define V4L2_PIX_FMT_RELI     v4l2_fourcc('R', 'E', 'L', 'I') /* 8 IR alternating on off illumination */

 /* SDR formats - used only for Software Defined Radio devices */
 #define V4L2_SDR_FMT_CU8          v4l2_fourcc('C', 'U', '0', '8') /* IQ u8 */`

but the second problem is:

`make: Entering directory '/home/firefly/workspace/librealsense/scripts/kernel/linux-4.4.114'
  Building modules, stage 2.
  MODPOST 0 modules
make: Leaving directory '/home/firefly/workspace/librealsense/scripts/kernel/linux-4.4.114'
cp: cannot stat '/home/firefly/workspace/librealsense/scripts/kernel/linux-4.4.114/drivers/media/usb/uvc/uvcvideo.ko': No such file or directory`

And I did'nt found the "$KBASE/drivers/media/usb/uvc/uvcvideo.ko" file.

Any one can help this ?

dorodnic, 2018-12-10:
Hi @dongxuanlb 
From my experience, RK Linux images tend to be monolithic (without Kernel modules) and I personally struggled to rebuild the entire kernel from source.
A more reasonable approach is to run CMake with flag `-DFORCE_LIBUVC=true`. Then everything should work.

dongxuanlb, 2018-12-10:
Hi @dorodnic 

I think maybe the problem is modpost does not build anything?

`firefly@firefly:~/workspace/librealsense/scripts/kernel/linux-4.4.114$ sudo make M=drivers/media/usb/uvc/ modules
  Building modules, stage 2.
  MODPOST 0 modules`

so there is no uvcvideo.ko in the folder?

dorodnic, 2018-12-10:
I'm not 100% sure, but the libuvc work-around should definitely work. @ev-mp will be back within a week and perhaps will be able to offer a better solution (I'm not an expert in kernel building :) )

RealSense-Customer-Engineering, 2019-01-15:
[Realsense Customer Engineering Team Comment]
Closing this one for now...
Please let us know if there is any further issue after enabling the libuvc workaround on your side.
Thank you.

closed 2019-01-15 by RealSense-Customer-Engineering