realsenseai/librealsense #2869: Retrieving intrinsic parameters after post-processing

shgold, 2018-12-10:
| Camera Model                       | { D400 } | 
| Operating System & Version |   {Linux 18.04} | 
| SDK Version                            |  { legacy / 2. }                          | 
| Language                            |  {python}                          | 

Hello, 
In the documentation, after post-processing the frames with 'decimation filter', the intrinsic parameters will be recalculated. 
I apply the filters to the frame however 'get_intrinsic()' function only works for the rs2::video_stream_profile object. How could I retrieve these new intrinsic parameters after post-processing?

dorodnic, 2018-12-10:
Hi @shgold 

```python
Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:19:30) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs2
>>> d = rs2.decimation_filter()
>>> p = rs2.pipeline()
>>> p.start()
<pyrealsense2.pyrealsense2.pipeline_profile object at 0x008DEEE0>
>>> fs = p.wait_for_frames()
>>> df = fs.get_depth_frame()
>>> processed = d.process(df)
>>> prof = processed.get_profile()
>>> video_prof = prof.as_video_stream_profile()
>>> intr = video_prof.get_intrinsics()
>>> intr
width: 640, height: 360, ppx: 316.603, ppy: 181.593, fx: 322.773, fy: 322.773, model: Brown Conrady, coeffs: [0, 0, 0, 0, 0]
```

shgold, 2018-12-10:
Thanks! It works!

closed 2018-12-10 by shgold