realsenseai/librealsense #2870: postprocessing

Mikeraka, 2018-12-10:
Hello, I want to set the depth unit

dorodnic, 2018-12-10:
Hi @Mikeraka 
Please take a look at [depth_filters.ipynb](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb), I think it can help you get started.

dorodnic, 2018-12-11:
Hi @Mikeraka 
Setting depth unit option should affect raw depth values. 
You can try it with the RealSense Viewer: when hovering over depth pixel, the reported distance should not change, but the raw value should.
Please check that you are using the latest SDK version, since we did fix something related to depth units recently.

Mikeraka, 2018-12-11:
@dorodnic Thank you very much for your answering.
As I want to save a sequences of aligned depth and rgb images for 3D reconstruction. My code is as follows:

import pyrealsense2 as rs
import numpy as np
import cv2
pipeline = rs.pipeline()

config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

decimation = rs.decimation_filter()
spatial = rs.spatial_filter()
temporal = rs.temporal_filter()
hole_filling = rs.hole_filling_filter()
depth_to_disparity = rs.disparity_transform(True)
disparity_to_depth = rs.disparity_transform(False)

profile = pipeline.start(config)

depth_sensor = profile.get_device().first_depth_sensor()

if depth_sensor.supports(rs.option.depth_units):
	depth_sensor.set_option(rs.option.depth_units,0.0001)#像素1——>代表0.0003m

depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

clipping_distance_in_meters = 1 #1 meter
clipping_distance = clipping_distance_in_meters / depth_scale

align_to = rs.stream.color
align = rs.align(align_to)

i=1
j=1
try:
    while True:
        frames = pipeline.wait_for_frames()
        
        aligned_frames = align.process(frames)
       
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = aligned_frames.get_color_frame()
        
        if not aligned_depth_frame or not color_frame:
            continue
        frame = depth_to_disparity.process(aligned_depth_frame)
        frame = spatial.process(frame)
        frame = temporal.process(frame)
        filtered_depth_frame = disparity_to_depth.process(frame)

        depth_image = np.asanyarray(filtered_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        
        grey_color = 153
        depth_image_3d = np.dstack((depth_image,depth_image,depth_image)) #depth image is 1 channel, color is 3 channels
        bg_removed = np.where((depth_image_3d > clipping_distance) | (depth_image_3d <= 0), grey_color, color_image)
        
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        images = np.hstack((bg_removed, depth_colormap))
        cv2.namedWindow('Align Example', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Align Example', images)
        cv2.imwrite("//home//cindy//0.0001WithLed//depth//"+str(i)+".png",depth_image)  
        cv2.imwrite("//home//cindy//0.0001WithLed//rgb//"+str(i)+".png",color_image) 
        i = i+1
        j = j+1
        key = cv2.waitKey(1)
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
finally:
    pipeline.stop()

I set the depth unit to 0.0001, the output Depth Scale is:  0.0010000000474974513. but I find a problem that in different lighting conditions for the same place, the depth value was totally different, when the light is strong(I turn on the light in the room), the depth value are among 1000 to 3000 or so. but when I turn off the light, for the same place in the same distance, the depth value are among 8000 to 10000 or so. It's really weird, what's the matter?
Thanks very much!

chaojizhangdaxin, 2018-12-13:
楼主，我想问一下使用Python怎么生成跟Windows的SDK生成的深度图片一样

RealSense-Customer-Engineering, 2018-12-14:
[Realsense Customer Engineering Team Comment]
@Chaojizhangdaxin Could you please provide a snapshot that you get from Windows SDK?

RealSense-Customer-Engineering, 2018-12-19:
[Realsense Customer Engineering Team Comment]
@Mikeraka Could you please provide some depth snapshots for your findings on the difference with different light conditions? Thanks!

RealSense-Customer-Engineering, 2019-01-03:
[Realsense Customer Engineering Team Comment]
@Mikeraka  Could you please update so we can follow up? Thanks!

Mikeraka, 2019-01-10:
@chaojizhangdaxin the depth image generated from Windows SDK is different from generated in python?

chaojizhangdaxin, 2019-01-10:
@Mikeraka My own program, the depth value error is relatively large. So I want to improve the accuracy of my own program.

closed 2019-01-14 by RealSense-Customer-Engineering