realsenseai/librealsense #2871: UV coordinates are zero sometimes

heavyman1, 2018-12-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.9.13 | 
| Operating System & Version |   Linux (Ubuntu 16.04.1)  | 
| Kernel Version (Linux Only)    |  4.4.0-138                                         | 
| Platform                                 | Up Board  |
| SDK Version                            |  2.12.0                          | 
| Language                            |  C++                         | 
| Segment			|  others                           | 

### Issue Description
Below code is part of my application.
In most cases it works fine.
Sometimes all UV coordinates but zero-z vertices are zero.
Is there any workaround?

	// Tell pointcloud object to map to this color frame
	rs2::pointcloud pc;
	pc.map_to(this->color_frame);

	// Generate the pointcloud and texture mappings
	rs2::points points = pc.calculate(filtered_depth_frame);

	// save vertices and uvs
	const rs2::vertex* rs_vertices = points.get_vertices();
	const rs2::texture_coordinate* rs_tex_coords = points.get_texture_coordinates();
	const int pointNumbers = (int)points.size();

	this->vertices.reserve(pointNumbers);
	this->uvs.reserve(pointNumbers);
	int zeroZNumbers = 0;
	int outOfTextureNumbers = 0;
	int borderTextureNumbers = 0;
	for (int vi = 0; vi < pointNumbers; vi++)
	{
		if ((IsZero(rs_vertices[vi].z)))
		{
			zeroZNumbers++;
		}
		else if ((rs_tex_coords[vi].u < 0.0f || 1.0f < rs_tex_coords[vi].u) ||
				(rs_tex_coords[vi].v < 0.0f || 1.0f < rs_tex_coords[vi].v))
		{
			outOfTextureNumbers++;
		}
		else if ((rs_tex_coords[vi].u == 0.0f || 1.0f == rs_tex_coords[vi].u) ||
				(rs_tex_coords[vi].v == 0.0f || 1.0f == rs_tex_coords[vi].v))
		{
			borderTextureNumbers++;
		}

		if ((IsNotZero(rs_vertices[vi].z)) &&
			(0.0f < rs_tex_coords[vi].u && rs_tex_coords[vi].u < 1.0f) &&
			(0.0f < rs_tex_coords[vi].v && rs_tex_coords[vi].v < 1.0f))
		{
			this->vertices.push_back(Vector3(rs_vertices[vi].x, rs_vertices[vi].y, rs_vertices[vi].z) * depthUnit);
			this->uvs.push_back(Vector2(rs_tex_coords[vi].u, rs_tex_coords[vi].v));
		}
	}
	this->vertices.shrink_to_fit();
	this->uvs.shrink_to_fit();

matkatz, 2018-12-11:
Hi,
There were major changes in the pointcloud filter since v2.12.0.
Can you please test your code again with the latest release - [v2.17.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.17.0)

heavyman1, 2018-12-11:
OK.
I will test again using 2.17.0 and FW 5.10.13

heavyman1, 2018-12-17:
After update library and fw, the problem was gone.
Thanks for your advice.

closed 2018-12-17 by heavyman1