realsenseai/librealsense #2876: How to use Python to get depth values on depth images

chaojizhangdaxin, 2018-12-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

chaojizhangdaxin, 2018-12-11:
My current program uses python, and I want to get depth values in the depth images I'm storing now.

RealSense-Customer-Engineering, 2018-12-13:
[Realsense Customer Engineering Team Comment]
The python-tutorial-1 provides simple example to capture and get the depth values, please give it try:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py

chaojizhangdaxin, 2018-12-14:
@RealSense-Customer-Engineering 

我以前看过这个实例，我没有看懂。我现在想的得到像素点（320,240）点的深度值多少米
I've seen this example before, but I don't understand it. What I want to get now is the depth of the pixels (320,240) in meters.

RealSense-Customer-Engineering, 2019-01-10:
[Realsense Customer Engineering Team Comment]
Hi chaojizhangdaxin,

If referring to the sample code, the depth value at any pixel point (x,y) can be obtained by

    dist = depth.get_distance(x, y)

shenzongyi1997, 2020-03-02:
Hi, I know how to get the distance of one single pixel. But I want to get the distance of all pixels at once, because when I use get_depth_image it will return me a ushort 2D array and its value is not distance. Is there any function like get_distance_image?

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

@shenzongyi1997 For the image distance you can refer to "Distance to Target" in the depth quality tool.

https://github.com/IntelRealSense/librealsense/tree/master/tools/depth-quality#distance-to-target

RealSenseCustomerSupport, 2020-03-17:
----------------------------------------------

@shenzongyi1997 Any other questions about this? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

@shenzongyi1997 Close it as no other questions. Thanks!

closed 2020-03-23 by RealSenseCustomerSupport

Ted-EE, 2020-07-20:
> @shenzongyi1997 For the image distance you can refer to "Distance to Target" in the depth quality tool.
> 
> https://github.com/IntelRealSense/librealsense/tree/master/tools/depth-quality#distance-to-target

Hi, I just met the same requirement. What I need to do is to remove the pixel whose distance is larger than 5 metres. However, it seems that using get_distance by traverse all pixels may take a bit long time and affect the performance of whole programme. Therefore, I would like to learn about if there exists a syntax which can show the distances of all pixels at once, so that I can remove the pixel whose distance is larger than 5 metres. 

I have referred your link Distance To Target, but I don't think ROI will help me solve this problem, only a syntax to achieve distances of all pixels may help.

Thanks