realsenseai/librealsense #2878: Add Hardware_Reset to .Net wrapper

smirkingman, 2018-12-11:
| Camera Model                       | All  | 
| Firmware Version                   | All | 
| Operating System & Version | Win (8.1/10)   | 
| Platform                                 | PC |
| SDK Version                            |  All | 
| Language                            |  C# | 

NativeMethods.cs declares rs2_hardware_reset but there is no corresponding call from managed code.

Presumably the implementation would be in Device.cs, this worked for me:

        public void HardwareReset()
        {
            object error;
            NativeMethods.rs2_hardware_reset(m_instance, out error);
        }

I understand that I could create a pull request but my C++ skills are poor and my Git skills non-existent, so it would be very appreciated if this was implemented by someone competent.

victorsbd, 2019-02-05:
https://github.com/IntelRealSense/librealsense/pull/3192/files/72f23b49d368dec0db914dd54ab53396dfea8e66

dorodnic, 2019-02-17:
Thanks @victorsbd

closed 2019-02-17 by dorodnic