realsenseai/librealsense #2880: depth with zero value

Usernamezhx, 2018-12-12:
I want to train a net with the depth image. but the depth image has the 0 value. it can cause the INF when i use the log function. how to deal with the zero value? thanks in advance.

MartyG-RealSense, 2018-12-12:
There was a recent case where a user had gaps in their depth data because some pixels were being registered as invalid (NaN) values instead of 0.  The approach they used to solve it was to write a script to convert the invalid values into zeroes.

 https://forums.intel.com/s/question/0D50P0000490XLwSAM/getting-complete-point-cloud-from-d435?language=en_US

Usernamezhx, 2018-12-13:
[image]
[image]
thanks for your reply. I think the light has an effect on the depth image. I will replace the 0's with a small value. But the left side of the depth image has an rectangle black region. I think I should cut it off.

closed 2018-12-13 by Usernamezhx