realsenseai/librealsense #2881: Depth image is not clear when picturing it at a far distance

Mikeraka, 2018-12-12:
Hello,I use the python code as follows to get aligned depth and rgb images, The problem is that  when I was near to the object in one meters, the depth image is clear, like figure1, but when I was about 2 to 3 meters to the object, the depth image was not clear, like figure2, why? is it mean that when I want to get depth image of something, I should not be far from it? I should take the depth image of it in 1 meter or so? or is it the setting's problem? how to set the sensor(D435) to get a more clear depth image of the object in a more far distance?

import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()

config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

profile = pipeline.start(config)

depth_sensor = profile.get_device().first_depth_sensor()

if depth_sensor.supports(rs.option.depth_units):
depth_sensor.set_option(rs.option.depth_units,0.001)

if depth_sensor.supports(rs.option.emitter_enabled):
depth_sensor.set_option(rs.option.emitter_enabled, 1.0)

depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

clipping_distance_in_meters = 1 #1 meter
clipping_distance = clipping_distance_in_meters / depth_scale

align_to = rs.stream.color
align = rs.align(align_to)

i=1
j=1
try:
while True:
frames = pipeline.wait_for_frames()
aligned_frames = align.process(frames)

    aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
    color_frame = aligned_frames.get_color_frame()
    
    if not aligned_depth_frame or not color_frame:
        continue
    
    depth_image = np.asanyarray(aligned_depth_frame.get_data())
    color_image = np.asanyarray(color_frame.get_data())
    
    grey_color = 153
    depth_image_3d = np.dstack((depth_image,depth_image,depth_image)) #depth image is 1 channel, color is 3 channels
    bg_removed = np.where((depth_image_3d > clipping_distance) | (depth_image_3d <= 0), grey_color, color_image)
    
    depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
    images = np.hstack((bg_removed, depth_colormap))
    cv2.namedWindow('Align Example', cv2.WINDOW_AUTOSIZE)
    cv2.imshow('Align Example', images)
    cv2.imwrite("//home//cindy//emitter_enabled//depth//"+str(i)+".png",depth_image)  #保存到指定目录
    cv2.imwrite("//home//cindy//emitter_enabled//rgb//"+str(i)+".png",color_image) 
    i = i+1
    j = j+1
    key = cv2.waitKey(1)
    if key & 0xFF == ord('q') or key == 27:
        cv2.destroyAllWindows()
        break
finally:
pipeline.stop()

figure1(in a close distance), the depth image is clear
[image]

figure2(in a far distance), the depth image of the object is not clear especially the outlier
[image]

Thanks very much

MartyG-RealSense, 2018-12-12:
As the distance from the camera increases, depth noise increases.  This is called RMS Error, and becomes particularly noticeable after 3 meters away or further.  This effects the D435 model particularly due to how its hardware is designed - the D415 has less RMS error at distance.

[image]

Intel's illustrated camera tuning guide may help you to improve the quality of the image.

https://realsense.intel.com/wp-content/uploads/sites/63/BKM-For-Tuning-D435-and-D415-Cameras-Webinar_Rev3.pdf

RealSense-Customer-Engineering, 2019-01-04:
[Realsense Customer Engineering Team Comment]
Closing this now.
 Feel free to reopen if the problem hasn't been addressed.
Thank you.

closed 2019-01-04 by RealSense-Customer-Engineering