realsenseai/librealsense #2885: Unable to measure depth for some pixels

ajprax, 2018-12-13:
<b>Camera Model</b>: D415
<b>Firmware Version</b>: 5.10.??? (can't check right now, but updated very recently)
<b>Operating System & Version:</b> MacOS 10.13.6
<b>Platform:</b> PC
<b>SDK Version:</b> 2.2.17
<b>Language:</b> python
<b>Segment:</b> Robot

I'm trying control the motion of a robot based on the 3d information generated by my camera. To do this, I need to calibrate a mapping between the coordinate system from the camera's perspective and the robot's. I've attached a marker to the robot and ordered it to move to various points, taking depth images at each point and getting the camera-space 3d point to build up the mapping.

The problem is that the camera is unable to determine the depth for my marker and therefor all the camera space points are coming back as (0, 0, 0). I see
>If a device fails to determine the depth of a given image pixel, a value of zero will be stored in the depth image.

in the wiki, but I'm wondering what conditions would cause a pixel to be unmeasurable. One case I've identified is if one of the two cameras can see something that the other cannot, like the side of an object, but my marker is facing straight up and should definitely be visible to both cameras.

What other cases can cause depth values to come out as 0? Are there ways to improve the camera's ability to measure these points.

MartyG-RealSense, 2018-12-13:
Are you using a flat surface such as a sticker for your marker, please?  And do you have an IR dot pattern projection from the camera active?  

There have been a couple of recent reports of users who were trying to track markers but failing.  They believed that the IR dot pattern projected by the camera was interfering with recognition of the marker.  

https://forums.intel.com/s/question/0D50P0000490YLVSA2/object-detection-in-ir-image-containing-projector-pattern

In that case, the user wished to use a 'Visual Preset' feature called 'Left Imager Color w/o IR Pattern' to remove the IR pattern from the stream.  This preset only works with D415, so as a D435 user, they were unable to use it.  As you have a D415 though, you could make use of this preset to see if it improves your marker detection.  If it doesn't work, then you will at least have eliminated the IR dot pattern as a potential cause of your detection problem.

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets

ajprax, 2018-12-13:
Thanks for the suggestions, Marty.

I am using a flat surface as the marker (highlighter on paper taped to the top face of the robot wrist).

`sensor.get_option(rs.option.emitter_enabled)` was on, but turning it off does not seem to have helped. I also tried `sensor.set_option(rs.option.visual_preset, 6)` for "Left Imager Color w/o IR Pattern" (along with a few other presets) with the same result.

A couple extra piece of information:
* The camera returns depth 0 for the top of the robot wrist with or without the marker
* The camera is facing straight down at a tabletop about 33cm away
* The top of the robot wrist is about 13cm from the camera
* Objects closer to the tabletop get pretty good depth measurements, including objects with the same marker

Is it possible the issue is with how close the objects are to the camera? From the "Related Discussion" on the visual presets wiki page I see some people talking about getting depth measurements for objects very close to the camera, so I'm going to keep reading there and see if any new ideas come up.

MartyG-RealSense, 2018-12-13:
The minimum depth distance (MinZ) of the D415 is listed as 0.3 m (30 cm).  So as long as your marker is more than 30 cm from the camera, it should be okay.  The image can deteriorate if the distance between marker and camera becomes less than 30 cm though.

You can reduce the MinZ minimum distance, at the cost of a shorter maximum distance (MaxZ), by changing the Disparity Shift value of the camera.  A simpler way to reduce MinZ is to reduce the resolution of the stream.

[image]

ajprax, 2018-12-13:
I don't see a "Disparity Shift" option in `rs.option`. I do see `min_distance` and `max_distance` but I get `Device does not support option Max Distance!` if I try to access them. This does seem like the most likely cause though, so if there's a way to set these values I'm hopeful things will work.

MartyG-RealSense, 2018-12-13:
Dorodnic the RealSense SDK Manager provides C++ code for changing disparity shift in the discussion linked to below:

https://github.com/IntelRealSense/librealsense/issues/2229

ajprax, 2018-12-14:
Tweaking the disparity shift has gotten me depth values for the marker, thanks!

Unfortunately, the minZ-maxZ range becomes quite small for higher disparity shift values, so I end up losing resolution at the surface. I think it's ok to use one disparity shift value to find the marker while calibrating and then lower the value to bring the surface back in range for normal operation.

MartyG-RealSense, 2018-12-14:
That makes sense.  Yes, others have sometimes found that using disparity shift was not appropriate for their particular application because of its side-effects.  So your balanced approach is a good idea.

ajprax, 2019-01-07:
I have a couple follow up questions.

1. Are these statements true?
   1. Disparity and disparity shift are in terms of pixels
   2. The disparity search range (126) cannot be changed
   3. The disparity for a particular depth increases when resolution increases
   4. Consequently, the size of the range of depths that can be measured effectively decreases as resolution increases
2. Is there a function available, either in code or as a mathematical expression, for determining the range of depths that can be read well given a resolution and camera attributes?

Thanks!

RealSense-Customer-Engineering, 2019-01-16:
[Realsense Customer Engineering Team Comment]
The min/maxZ and the properties of the disparity is also available in the following white paper:
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

closed 2019-01-24 by ajprax