realsenseai/librealsense #2887: What can I do to 3D Reconstruction in Unity?

GameHuiTea, 2018-12-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------
Required Info |  
-- | --
Camera Model | { D415 }
Firmware Version | (Open RealSense Viewer --> Click info)
Operating System & Version | {Win (10)
Kernel Version (Linux Only) |  
Platform | PC
SDK Version | { 2.0 (build 2.16.5) }
Language | { C#(unity) }
Segment | { }

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello, First of all, I'm sorry for asking so many questions. I think I should probably ask a lot of questions on this issue.

first I worked with the goal of accumulating the screen displayed as a point cloud without updating it.

To implement this, I thought I had to create a new object for each texture, uv and put on a new material that wasn't updated.

I don't know how to make sure the old Texture and UV aren't updated on the material I copied.
Code in Python:
`                GameObject afterImageObject = Instantiate(gameObject, Vector3.zero, Quaternion.identity);
                afterImageObject.GetComponent<MeshRenderer>().material = meshRenderer.material;
                afterImageObject.GetComponent<CRsPointCloudRenderer>().enabled = false;`

Let me know if you know any other way or if you think of keywords.
I will study hard and distribute information here.

Thank you for reading!
Have a nice day

HippoEug, 2018-12-13:
We need 3D reconstruction in the SDK so desperately! haha!

MartyG-RealSense, 2018-12-13:
There was recently a discussion about doing a point cloud to mesh conversion within the SDK, but the discussion's page link got broken when Intel Support's forums moved to a new system this week.

My recollection is that the user in that case achieved a mesh by switching the point cloud rendering from point cloud to quads.  This can be toggled in the RealSense Viewer, but this user also posted a short point cloud script for doing the conversion.  Unfortunately the discussion cannot be accessed.

It was almost the same as the standard demo script for generating a point cloud but slightly different - I believe that they used a different instruction to replace 'depth_frame' in the bracket on the second line.  Apologies that I cannot remember it better - hopefully one of the Intel guys can provide the code for rendering as quads.

rs2::pointcloud pc;
rs2::points points = pc.calculate(depth_frame);
pc.map_to(color_frame);

dorodnic, 2018-12-16:
Hi @GameHuiTea @HippoEug @MartyG-RealSense 
I'm not sure how this can be best accomplished in Unity, but there are some developments in this direction:
a. @doronhi just published a [draft of environment mapping using D435i](https://github.com/intel-ros/realsense/wiki/SLAM-with-D435i) with ROS
b. @AnnaRomanov just submitted a [PR](https://github.com/IntelRealSense/librealsense/pull/2903) showing how to run [Kinnect-Fussion](https://www.microsoft.com/en-us/research/wp-content/uploads/2016/02/ismar2011.pdf) with RealSense. (see [the new example](https://github.com/AnnaRomanov/librealsense/tree/kinfu-example/wrappers/opencv/kinfu))
c. We have several [software partners](https://realsense.intel.com/software-for-intel-realsense/) who specialise in 3D scanning

HippoEug, 2018-12-17:
Amazing. Looking forward for some sweet SDK integration!

closed 2019-04-07 by dorodnic