realsenseai/librealsense #2888: Decopuple the logic to process frames from the UI.

eebanos, 2018-12-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 } | 
| Firmware Version                   | (3.21.0.0) | 
| Operating System & Version |   {Win (10)  |                                      | 
| Platform                                 | PC |
| SDK Version                            |  { 2.16.3.0 }                          | 
| Language                            |  {C# }                          | 
| Segment			|  {others }                          | 

### Issue Description

Following the example, specifically this one: [cs-tutorial-3-processing](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp/cs-tutorial-3-processing) I'm trying to decouple the logic to be able to display the frame result from a WPF Page, through a property defined inside a controller class implementing the INotifyPropertyChanged interface.
The main issue is how do I access a the ColorFrame and display it without using the Dispatcher.Invoke approach.

RealSense-Customer-Engineering, 2019-01-08:
[Realsense Customer Engineering Team Comment]
Hi eebanos,

I am not so familiar with C#, but would like to give it a try...
If I understand correctly, and then take into a more generic way on separating GUI from inside business logic.
If this is the case, (with some searching on the internet), you may want to take a look into MVC pattern's variation MVVM Pattern

Thank you!

eebanos, 2019-01-10:
The MVVM pattern is the one I'm using, solved in combination with the async/await using sequencial tasks for post-processing the frames.

closed 2019-01-10 by eebanos

RealSense-Customer-Engineering, 2019-01-10:
[Realsense Customer Engineering Team Comment]
Hi eebanos,

Wonder if you get update and still need help on this one? If not, this will be closed soon.

Thanks!

eebanos, 2019-01-16:
Already solved and closed, thanks.

RealSenseCustomerSupport, 2019-01-21:
Thanks for the update, and glad to know that you solved it with MVVM.

BTW, there was some issue with our system and some of the messages were not updated timely in our end.

Thanks!