realsenseai/librealsense #2891: Android, D415, rs2:backend_error, Permission Denied

adraffy, 2018-12-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415| 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |  MacOS 10.14.1 / Android 7.1.1  | 
| Platform                                 | PC/Android  |
| SDK Version                            |  2 }                          | 
| Language                            |  Java/C                          | 
| Segment			|  Tablet? }                          | 

### Issue Description

I have a rooted Lenovo Tab 4 8" running Android 7.1.1

I have successfully compiled the [Android version of the project](https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidJavaApp.md). 

I have connected the camera to my tablet using two cables:
- USB mini A to USB A (UGREEN Micro USB to USB, Micro USB 2.0 OTG Cable 2 Pack On The Go Adapter Micro USB Male to USB Female)
- USB A to USB C (supplied with D415)

When I launch "realsense_app", it correctly detects the presence of the camera (ie. error if usb is not connected.)  However, quickly after that, the application crashes:
```
 signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
2018-12-13 13:26:11.563 12956-12956/? A/DEBUG: Abort message: '/Volumes/Android/buildbot/src/android/ndk-release-r18/external/libcxx/../../external/libcxxabi/src/abort_message.cpp:73: abort_message: assertion "terminating with uncaught exception of type rs2::backend_error: Could not open device. Last Error: Permission denied" failed'
2018-12-13 13:26:11.563 12956-12956/? A/DEBUG:     r0 00000000  r1 00003261  r2 00000006  r3 00000008
2018-12-13 13:26:11.563 12956-12956/? A/DEBUG:     r4 d8460978  r5 00000006  r6 d8460920  r7 0000010c
2018-12-13 13:26:11.563 12956-12956/? A/DEBUG:     r8 ffc3769c  r9 f03db180  sl e9b06929  fp 0000000b
2018-12-13 13:26:11.563 12956-12956/? A/DEBUG:     ip 0000000d  sp d8460220  lr f25d6467  pc f25d8ce8  cpsr 200e0010
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG: backtrace:
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #00 pc 00049ce8  /system/lib/libc.so (tgkill+12)
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #01 pc 00047463  /system/lib/libc.so (pthread_kill+34)
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #02 pc 0001d745  /system/lib/libc.so (raise+10)
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #03 pc 00019281  /system/lib/libc.so (__libc_android_abort+34)
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #04 pc 000172e8  /system/lib/libc.so (abort+4)
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #05 pc 0001b73f  /system/lib/libc.so (__libc_fatal+22)
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #06 pc 0001947b  /system/lib/libc.so (__assert2+18)
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #07 pc 0082e6e9  /data/app/com.example.realsense_app-1/lib/arm/librealsense2.so
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #08 pc 0082e7f5  /data/app/com.example.realsense_app-1/lib/arm/librealsense2.so
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #09 pc 0082cb95  /data/app/com.example.realsense_app-1/lib/arm/librealsense2.so
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #10 pc 0082c7a7  /data/app/com.example.realsense_app-1/lib/arm/librealsense2.so (__cxa_rethrow+78)
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #11 pc 0000adaf  /data/app/com.example.realsense_app-1/lib/arm/libnative-lib.so
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #12 pc 0000a9e3  /data/app/com.example.realsense_app-1/lib/arm/libnative-lib.so
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #13 pc 00046f33  /system/lib/libc.so (_ZL15__pthread_startPv+22)
2018-12-13 13:26:11.573 12956-12956/? A/DEBUG:     #14 pc 00019ccd  /system/lib/libc.so (__start_thread+6)
```

With the primary error being the following exception: 
```
rs2::backend_error: Could not open device. Last Error: Permission denied" failed'
```

I traced this error to the function `power_D0()` inside of `libuvc/libuvc.cpp`.  It looks like `uvc_find_device()` returns 0, so then it calls `uvc_open2` with `interface=0` and returns `res=-3` causing the following exception to be thrown.  Error `-3` appears to be `LIBUSB_ERROR_ACCESS` from `libusb_open`which corresponds to "Access denied (insufficient permissions)".

I've checked that the USB device permissions are okay:  when my camera is connected, I see 2 USB devices, 001/001 and 001/002, and I've chmod -R 0777 the appropriate files.

Any ideas?

adraffy, 2018-12-13:
Prior to the exception, I am seeing the following messages:
```
2018-12-13 14:56:47.430 18497-18497/com.example.realsense_app I/e.realsense_app: type=1400 audit(0.0:333): avc: denied { read } for name="usb" dev="tmpfs" ino=56233 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:usb_device:s0 tclass=dir permissive=1
2018-12-13 14:56:47.430 18497-18497/com.example.realsense_app I/e.realsense_app: type=1400 audit(0.0:334): avc: denied { open } for path="/dev/bus/usb" dev="tmpfs" ino=56233 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:usb_device:s0 tclass=dir permissive=1
2018-12-13 14:56:47.430 18497-18497/com.example.realsense_app I/e.realsense_app: type=1400 audit(0.0:335): avc: denied { create } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:untrusted_app:s0:c512,c768 tclass=netlink_kobject_uevent_socket permissive=1
2018-12-13 14:56:47.430 18497-18497/com.example.realsense_app I/e.realsense_app: type=1400 audit(0.0:336): avc: denied { bind } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:untrusted_app:s0:c512,c768 tclass=netlink_kobject_uevent_socket permissive=1
2018-12-13 14:56:47.430 18497-18497/com.example.realsense_app I/e.realsense_app: type=1400 audit(0.0:337): avc: denied { setopt } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:untrusted_app:s0:c512,c768 tclass=netlink_kobject_uevent_socket permissive=1
2018-12-13 14:56:47.430 18497-18497/com.example.realsense_app I/e.realsense_app: type=1400 audit(0.0:338): avc: denied { read } for name="devices" dev="sysfs" ino=15933 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1
2018-12-13 14:56:47.430 18497-18497/com.example.realsense_app I/e.realsense_app: type=1400 audit(0.0:339): avc: denied { open } for path="/sys/bus/usb/devices" dev="sysfs" ino=15933 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1
2018-12-13 14:56:47.430 18497-18497/com.example.realsense_app I/e.realsense_app: type=1400 audit(0.0:340): avc: denied { read } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:untrusted_app:s0:c512,c768 tclass=netlink_kobject_uevent_socket permissive=1
```

adraffy, 2018-12-14:
Not sure what changed but after restarting the Android device, replugging, and applying permissions again, I can now get the app to run for about 5-10 seconds until it SIGSEGVs.

```
2018-12-13 16:22:28.655 13250-13250/? A/DEBUG: Build fingerprint: 'Lenovo/TB-8504F/TB-8504F:7.1.1/NMF26F/TB-8504F_S000061_180912_ROW:user/release-keys'
2018-12-13 16:22:28.655 13250-13250/? A/DEBUG: Revision: '0'
2018-12-13 16:22:28.655 13250-13250/? A/DEBUG: ABI: 'arm'
2018-12-13 16:22:28.655 13250-13250/? A/DEBUG: pid: 13180, tid: 13221, name: Thread-4  >>> com.example.realsense_app <<<
2018-12-13 16:22:28.655 13250-13250/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xd6d80000
2018-12-13 16:22:28.655 13250-13250/? A/DEBUG:     r0 d6d3e200  r1 d6d80000  r2 000043c0  r3 00000000
2018-12-13 16:22:28.655 13250-13250/? A/DEBUG:     r4 d6d44f80  r5 e74e8198  r6 d7e7e948  r7 d7e7e908
2018-12-13 16:22:28.655 13250-13250/? A/DEBUG:     r8 00000000  r9 5a313620  sl 0000001e  fp 0000010e
2018-12-13 16:22:28.655 13250-13250/? A/DEBUG:     ip f48a1624  sp d7e7e8b8  lr db0da773  pc f483476c  cpsr 20010010
2018-12-13 16:22:28.662 13250-13250/? A/DEBUG: backtrace:
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #00 pc 0001776c  /system/lib/libc.so (memcpy+96)
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #01 pc 006ef76f  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so (_ZN12librealsense4copyEPvPKvj+92)
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #02 pc 0068f0d9  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so (_ZN12librealsense11copy_pixelsILj2EEEvPKPhPKhii+52)
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #03 pc 0065e2d9  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #04 pc 0065d411  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #05 pc 0065d03d  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #06 pc 00489f2b  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so (_ZNKSt6__ndk18functionIFvN12librealsense8platform14stream_profileENS2_12frame_objectENS0_IFvvEEEEEclES3_S4_S6_+190)
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #07 pc 0055e6ef  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so (_ZN12librealsense8platform17libuvc_uvc_device12uvc_callbackEP9uvc_frameNSt6__ndk18functionIFvNS0_14stream_profileENS0_12frame_objectENS5_IFvvEEEEEES6_+214)
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #08 pc 0055e3e7  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #09 pc 0055439f  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so (_Z16_uvc_user_callerPv+374)
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #10 pc 00554ecf  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #11 pc 00554dd3  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #12 pc 00046f33  /system/lib/libc.so (_ZL15__pthread_startPv+22)
2018-12-13 16:22:28.663 13250-13250/? A/DEBUG:     #13 pc 00019ccd  /system/lib/libc.so (__start_thread+6)
```

FYI: There is definitely a bug with the ARGB conversion code in `onFrame`.  In Java, byte needs unsigned before it is shifted (eg. ((byte)-1)<<8 =!= (-1&0xFF)<<8.)

dorodnic, 2018-12-16:
Couple of ideas:
1. Firmware update can help (05.08.15.00 is one year old, and there has been a lot of firmware fixes and improvements since then)
2. You could try reducing the resolution
3. Enable YUY stream instead of RGB and do the conversion in Render Script to reduce CPU utilization

adraffy, 2018-12-19:
Ah, thank you, the lagging camera firmware was the culprit.

adraffy, 2018-12-20:
I've still got lots of issues.  I guess I'll just detail them here instead of creating a new issue.

**1.)** I frequently get random crashes of the following on the first call to `pipe.wait_for_frames()`:
```
2018-12-19 14:22:14.639 31892-31937/? A/libc: Fatal signal 11 (SIGSEGV), code 1, fault addr 0xd7580000 in tid 31937 (Thread-4)
2018-12-19 14:22:14.713 31939-31939/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2018-12-19 14:22:14.713 31939-31939/? A/DEBUG: Build fingerprint: 'Lenovo/TB-8504F/TB-8504F:7.1.1/NMF26F/TB-8504F_S000061_180912_ROW:user/release-keys'
2018-12-19 14:22:14.713 31939-31939/? A/DEBUG: Revision: '0'
2018-12-19 14:22:14.713 31939-31939/? A/DEBUG: ABI: 'arm'
2018-12-19 14:22:14.714 31939-31939/? A/DEBUG: pid: 31892, tid: 31937, name: Thread-4  >>> com.example.realsense_app <<<
2018-12-19 14:22:14.714 31939-31939/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xd7580000
2018-12-19 14:22:14.714 31939-31939/? A/DEBUG:     r0 d747c400  r1 d7580000  r2 00019bc0  r3 00000000
2018-12-19 14:22:14.714 31939-31939/? A/DEBUG:     r4 d7503c00  r5 e74e8770  r6 d787e948  r7 d787e908
2018-12-19 14:22:14.714 31939-31939/? A/DEBUG:     r8 00000000  r9 5a313620  sl 00000006  fp 000001e0
2018-12-19 14:22:14.714 31939-31939/? A/DEBUG:     ip f48a1624  sp d787e8b8  lr db0da773  pc f483476c  cpsr 20010010
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG: backtrace:
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #00 pc 0001776c  /system/lib/libc.so (memcpy+96)
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #01 pc 006ef76f  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so (_ZN12librealsense4copyEPvPKvj+92)
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #02 pc 0068f0d9  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so (_ZN12librealsense11copy_pixelsILj2EEEvPKPhPKhii+52)
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #03 pc 0065e2d9  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #04 pc 0065d411  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #05 pc 0065d03d  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #06 pc 00489f2b  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so (_ZNKSt6__ndk18functionIFvN12librealsense8platform14stream_profileENS2_12frame_objectENS0_IFvvEEEEEclES3_S4_S6_+190)
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #07 pc 0055e6ef  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so (_ZN12librealsense8platform17libuvc_uvc_device12uvc_callbackEP9uvc_frameNSt6__ndk18functionIFvNS0_14stream_profileENS0_12frame_objectENS5_IFvvEEEEEES6_+214)
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #08 pc 0055e3e7  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #09 pc 0055439f  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so (_Z16_uvc_user_callerPv+374)
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #10 pc 00554ecf  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #11 pc 00554dd3  /data/app/com.example.realsense_app-2/lib/arm/librealsense2.so
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #12 pc 00046f33  /system/lib/libc.so (_ZL15__pthread_startPv+22)
2018-12-19 14:22:14.721 31939-31939/? A/DEBUG:     #13 pc 00019ccd  /system/lib/libc.so (__start_thread+6)
```

**2.)** Even though I explicitly only enable the stream(s) I want, librealsense will often give me frames of a previous resolution/fps setting until I unplug the camera.  For example, if I pull some 640x480 frames, then restart my app using 480x270, I'll often get mangled depth frames that are actually 640x480 but wrapped as 480x270.

**3.)** I am unable to pull color frames from the camera.  There is no combination of width/height/fps that seems to allow me to enable both two streams that don't result in "Unable to resolve requests."
```
rs2::config config;
config.disable_all_streams();
config.enable_stream(RS2_STREAM_COLOR, __, __, RS2_FORMAT_BGR8, __);
config.enable_stream(RS2_STREAM_DEPTH, __, __, RS2_FORMAT_Z16, __);
pipe.start(config);
```

dorodnic, 2018-12-20:
1 and 2 might be related - if unpacker gets wrong resolution, this can easily result in a SEGFAULT.
I don't think it was stated anywhere in the issue, what version of LRS are you using?

Regarding 3 - is this with the new firmware? Does the tablet provide USB3 port or just the regular USB2 OTG? I'm concerned that the camera might be detected as USB2 device.

RealSense-Customer-Engineering, 2019-01-09:
[Realsense Customer Engineering Team Comment]
@adraffy What's the FW version of your camera? If not the latest, could you please update to try again?
FW download link: https://downloadcenter.intel.com/download/28237/Latest-Firmware-for-Intel-RealSense-D400-Product-Family

adraffy, 2019-01-10:
@RealSense-Customer-Engineering I am using 5.10.13

RealSense-Customer-Engineering, 2019-01-14:
[Realsense Customer Engineering Team Comment]
@adraffy Did you get chance to try the latest FW? Any update? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-01-18:
@adraffy If you can apply kernel patch, please have a try v4l2 backend for librealsense as PR#2489
Or please wait some more days for the non-root solution.

RealSenseCustomerSupport, 2019-01-22:
@adraffy Did you get chance to try v4l2 backend solution(which need apply kernel patch)? Or you want to wait for some days for the without root solution?

adraffy, 2019-01-24:
Thanks for checking-in.  No, I haven't had time to test it yet.  I'll probably wait for the rootless solution to reinvestigate.

RealSenseCustomerSupport, 2019-01-24:
@adraffy OK. Thank you for your patience! Let's wait to see the rootless solution.

RealSenseCustomerSupport, 2019-02-21:
----------------------------------------------

@adraffy Please have a try with https://github.com/IntelRealSense/librealsense/pull/3083 . And please noted it was just merged to development branch and not officially publish yet.

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

@adraffy Any good luck with https://github.com/IntelRealSense/librealsense/pull/3083 ? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-02-28:
----------------------------------------------

@adraffy Any update? Thanks!

adraffy, 2019-02-28:
@RealSenseCustomerSupport I am still waiting on additional funding to resume work on this project.  I haven't been able to test.   

Thanks for your help.

closed 2019-02-28 by adraffy