realsenseai/librealsense #2894: Does the D435 camera need calibration? Distance of visual recognition items within 5 meters

chaojizhangdaxin, 2018-12-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu16/ | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            | 2.16.1                     | 
| Language                            |  opencv/python }                          | 
| Segment			|  Robo                        | 

### Issue Description
Do you need camera coordinates after aligning depth photographs with RGB photographs?

dorodnic, 2018-12-16:
Hi @chaojizhangdaxin 
Could you please provide more information on the issue? I'm not sure I fully understand.

* D400 would not be a very good solution at 5 meters. 
* Depth and Color cameras are calibrated and the SDK offers `align` class to align images.
* There are also APIs to get the relative transforms between the two infrared sensors and the RGB camera (see [camera extrinsics](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#get-and-apply-depth-to-color-extrinsics))

chaojizhangdaxin, 2018-12-17:
@dorodnic Thank you for your answer. I want to know whether the pixels of RGB and depth images correspond to each other point to point. For example, I want to know the depth value of (x, y) in RGB photos. Is that the depth value of (x, y) in depth images?

RealSense-Customer-Engineering, 2019-01-08:
[Realsense Customer Engineering Team Comment]
Hi chaojizhangdaxin,

As dorodnic and HippoEug clarified & suggested, please refer to align sample.

Thank you!

RealSense-Customer-Engineering, 2019-01-10:
[Realsense Customer Engineering Team Comment]
Hi chaojizhangdaxin,

Wonder if you get any updates and still need help? If no, this one will be closed soon.

Thanks!

closed 2019-06-03 by dorodnic