realsenseai/librealsense #2895: Support for kernels v4.16 and USB 3.1 in PC

sudeepbanik, 2018-12-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | Open RealSense  | 
| Operating System & Version |Linux (Ubuntu 14)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.16.0)                                         | 
| Platform                                 | PC |
| SDK Version                            | 1.12.0                         | 
| Language                            |  python}                          | 
| Segment			|  others }                          | 

### Issue Description

BlasterX Senz3D(SR300) camera won't connect.

Error message:

terminate called after throwing an instance of 'rs:: error'
what(): VIDIOC_S_CTRL error 25, Inappropriate ioctl for device
Aborted (core dumped)

### Note
Since my PC doesn't have a USB 3.0 port, I am using a USB to Type C converter, since my PC has a Type C 3.1 port.

MartyG-RealSense, 2018-12-14:
Kernel 4.16 has been supported in the SDK since version 2.16.1.  

It sounds as though your PC has what is known as a USB 3 OTG (On The Go) port, which is a micro-USB port that allows the attachment of full size USB devices via an adapter.

Is your PC a device such as a small single-board computer, please?

sudeepbanik, 2018-12-14:
I am using a laptop. My system configuration is:

**Model**: Lenovo Ideapad 320-15isk 80XH.
**Processor**: Intel Core i3-6006U (6th Gen).
**USB 2.0**:  2  Ports
**USB 3.0**: Not Available
**Type C USB 3.1**:  1 Port

I'm using a USB Type C to USB 3.0 converter to connect the the SR300 camera as I don't have any USB 3.0 port.

MartyG-RealSense, 2018-12-14:
Your processor is compatible with SR300, so the problem is not likely to be processor-related.  I would be inclined to think it was a USB issue.  The SR300 can have problems with full USB 3.0 ports in regard to stability, so a micro-USB port may introduce even more potential for problems (since they are susceptible to not delivering enough power to the camera).

If it is a USB related issue, using a mains-powered USB 3 hub with a USB-C connector to provide USB 3 ports for your laptop, and attaching the SR300 to the hub, may increase stability. 

https://www.amazon.com/s/ref=nb_sb_noss?url=search-alias%3Daps&field-keywords=usb+type-c+hub+powered

sudeepbanik, 2018-12-14:
I am using a USB Type-C to USB 3.1 Gen1 Adapter to connect my camera to the laptop but it still got the same error.

[https://www.amazon.com/AmazonBasics-Type-C-Gen1-Female-Adapter/dp/B01GGKYXVE/ref=sr_1_3?ie=UTF8&qid=1544783222&sr=8-3&keywords=usb+type-c+to+usb+3.1](url)

MartyG-RealSense, 2018-12-14:
It seems to be a 'passive' connector that draws power from the computer's USB port.  A mains-powered hub is 'active' because the power is supplied to the USB port from the mains socket supply rather than the PC's power supply unit.

ev-mp, 2018-12-16:
Hello @sudeepbanik ,
Librealsense version 1.12.0 (SDK1) is officially supported with LTS kernel 4.4.0-xxx. De-facto, we relaxed the constrain by allowing v1.12 to work correctly with all Ubuntu Xenial LTS kernels 4.[8/10/13/15] using SDK2's patches suite.
Kernel 4.16 introduced incompatible uvcvideo change by splitting the each video device into video and metadata nodes for native metadata support. This modification is [integrated exclusively](https://github.com/IntelRealSense/librealsense/pull/2275) into SDK 2.

Basically the options are:
 -- Upgrade the library to SDK version 2  ([2.17.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.17.0) is the latest), which has full support for SR300.
  -- Remain the Librealsense v1.12.0 , but downgrade the kernel to one of the [official Ubuntu LTS kernels](https://wiki.ubuntu.com/Kernel/LTSEnablementStack)

sudeepbanik, 2019-01-11:
Hi @RealSense-Customer-Engineering 
There is no issue in the SDK 2.17.0, but since I am required to do my work in SDK 1.12.0, I cannot upgrade my SDK to SDK 2.17.0. It is still causing a problem in SDK 1.12.0.

RealSense-Customer-Engineering, 2019-01-14:
[Realsense Customer Engineering Team Comment]
Closing it for now. Please reopen if there is any further question. Thank you.

closed 2019-01-14 by RealSense-Customer-Engineering