realsenseai/librealsense #2896: CANNOTLINKEXECUTABLE "./rs-depth': library "librealsense2.so" not found Aborted run in Android

marymsun, 2018-12-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |5.10.06 | 
| Operating System & Version |   Android8.1  | 
| Kernel Version (Linux Only)    |  4.4                                        | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  2.16.1 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Follow Intel-RealSense-SDK2-Github-Guide Android* Installation steps,when run ./rs-depth meet error with "CANNOTLINKEXECUTABLE "./rs-depth': library "librealsense2.so" not found Aborted"

How to fix this issue?Thanks!

[image]

Add logs with android platform
[image]

dorodnic, 2018-12-16:
Hi @marymsun 
[AndroidNativeSamples.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidNativeSamples.md) recommends to build with ` -DBUILD_SHARED_LIBS=false`. Please make sure you are passing this flag to CMake, and it should resolve this issue.

marymsun, 2018-12-17:
It works normal.Thanks!

closed 2018-12-18 by RealSense-Customer-Engineering