realsenseai/librealsense #2897: Third-Party broken

2er0, 2018-12-14:
third-party - lib issue
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | - | 
| Operating System & Version | Linux | 
| Kernel Version (Linux Only)    |  4.19.4                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.16.5                          | 
| Language                            |  C++                           | 
| Segment			|  others                          | 

### 1 Build Issue

after switching from version 2.16.4 to 2.16.5, I now get this compiling errors

```
In file included from ../librealsense/third-party/libtm/libtm/src/Device.cpp:21:
../librealsense/third-party/libtm/libtm/src/CentralAppFw.h:1347:2: error: #endif without #if
 #endif
  ^~~~~
In file included from ../git/librealsense/third-party/libtm/libtm/src/Device.cpp:22:
../librealsense/third-party/libtm/libtm/src/CentralBlFw.h:850:2: error: #endif without #if
 #endif
  ^~~~~
In file included from ../librealsense/third-party/libtm/libtm/src/Device.cpp:21:
../librealsense/third-party/libtm/libtm/src/CentralAppFw.h:682:1: error: expected unqualified-id before numeric constant
 0x00008902,0x00A4FC00,0x00000000,0x00000000,0x00000000,0x00000000,0x01861200,0x0108400A,0x93083C12,0x34100880,0x0198021A,0x00280020,0xFC380030,0x244202C9,0x20120308,0xC4772127,
 ^~~~~~~~~~
../librealsense/third-party/libtm/libtm/src/CentralAppFw.h:1345:1: error: expected declaration before ‘}’ token
 };
 ^
gmake[2]: *** [third-party/libtm/libtm/src/CMakeFiles/tm.dir/build.make:76: third-party/libtm/libtm/src/CMakeFiles/tm.dir/Device.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from ../librealsense/third-party/libtm/libtm/src/Manager.cpp:12:
../librealsense/third-party/libtm/libtm/src/fw.h:251221:2: error: #endif without #if
 #endif
  ^~~~~
../librealsense/third-party/libtm/libtm/src/fw.h:125617:1: error: expected unqualified-id before numeric constant
 0x7832414D,0x00001BA8,0x49726464,0x4874696E,0x65646165,0x726F4672,0x6D63764D,0x6C652E64,0x289A0066,0x1120FC00,0x9A111111,0x20FC0030,0x11111111,0x000012A8,0x435F3843,0x32335F30,
 ^~~~~~~~~~
../librealsense/third-party/libtm/libtm/src/fw.h:251220:1: error: expected declaration before ‘}’ token
 };
```
----------------------------------------------------------------------------------------------------
### 2 Build fix for another issue
Second in the file `third-party/libtm/resources/CMakeLists.txt` are some issues. 
The `string(REGEX REPLACE "..." "..." buffer ${buffer})` regex and replace need 6 parameters. 
There are 6, but the last one hast to be written in that way `"${buffer}"`.

dorodnic, 2018-12-16:
A quick work-around is to build with `-DBUILD_WITH_TM2=false`. 
However, this is a bit surprising since we do build it in our CI. 
Could you share the compiler version you are using?

abernste, 2018-12-16:
@2er0
1. If you download a clean 2.16.4 - cmake+compilation works on the source?
2. If you download a clean 2.16.5 - cmake+compilation works on the source?
3. If you download a clean 2.16.4, run cmake + compile and then update to 2.16.5 - does it fail with above error?
4. Regarding the REGEX issue - where do you see that it needs to be written with apostrophes (https://cmake.org/cmake/help/latest/command/string.html#regex-replace)
  Why do you think its not working without apostrophes?

Thanks for your help!

2er0, 2018-12-20:
@abernste 
#4: If I run CMake in release mode with CLion, I get this error:
```
-- Downloading Central BL 1.0.1.112 from 'http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/bl/1.0.1.112/central_bl-1.0.1.112.bin'
-- Downloading Central BL 1.0.1.112 completed - status 0;"No error"
-- Converting Central BL version 1.0.1.112 /home/dba/git/librealsense/third-party/libtm/resources/central_bl.bin to .../librealsense/third-party/libtm/libtm/src/CentralBlFw.h
-- Creating HEX buffer from .../librealsense/third-party/libtm/resources/central_bl.bin offset 3
CMake Error at third-party/libtm/resources/CMakeLists.txt:71 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
Call Stack (most recent call first):
  third-party/libtm/resources/CMakeLists.txt:290 (bin2h)
```

dorodnic, 2018-12-29:
Thank you @2er0 
I believe this was address as part of [2940](https://github.com/IntelRealSense/librealsense/pull/2940/files)

closed 2019-02-04 by dorodnic