realsenseai/librealsense #2899: How to find devices with motion sensors (IMU)?

UnaNancyOwen, 2018-12-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| SDK Version                            |  2.17.0                          | 
| Language                            |  C++                          | 

### Issue Description
Do you know how to find devices with motion sensors (IMU)?
I think it is able to do that, but it require to write a many lines code. I want an easy way.
If there is no easy way to do that, I suggest adding a mask to <code>rs2::context::query_devices()</code>.

```cpp
rs2::context;
context.query_devices( RS2_PRODUCT_LINE_D400 | RS2_HAVE_IMU ); // return d435i devices
```

UnaNancyOwen, 2018-12-16:
Yes, It will be able to identify RealSense D435i. However, It is hard coding.
It will be troublesome when another RealSense with IMU is released in the future.
Is there a way to not require hard coding?

```cpp
// The example that I showed was misleading. This is what I want to do.
rs2::context;
const rs2::device_list device_list = context.query_devices( RS2_PRODUCT_LINE_ANY_INTEL | RS2_HAVE_IMU ); // return devices with imu
```

UnaNancyOwen, 2018-12-16:
This is what I want to do. However, I want to write it more easily.

```cpp
void query_motion_devices( std::vector<rs2::device>& motion_devices )
{
    rs2::context context;
    const rs2::device_list device_list = context.query_devices( RS2_PRODUCT_LINE_ANY_INTEL );
    for( const rs2::device& device : device_list ){
        const std::vector<rs2::sensor> sensors = device.query_sensors();
        for( const rs2::sensor& sensor : sensors ){
            const std::vector<rs2::stream_profile> profiles = sensor.get_stream_profiles();
            for( const rs2::stream_profile& profile : profiles ){
                if( profile.is<rs2::motion_stream_profile>() ){
                    motion_devices.push_back( device );
                }
            }
        }
    }
}
```

MartyG-RealSense, 2018-12-16:
I would imagine that the IMU component has some kind of individual recognition code so that a program can detect whether the sensor is present (like with the 'sensor-control' sample program that lists the sensor details).

sensor-control iterates through the available sensors on a camera suing a command called get_sensor_name.  You could probably modify the terms of the search to only react positively to the name of the IMU sensor component (which would presumably be the same between different cameras using the same IMU component) and not allow the program to run if the search condition does not detect the sensor name.

https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/api_how_to.h#L131

dorodnic, 2018-12-16:
Hi @UnaNancyOwen 
Yes, we need to add this (either via PRODUCT_LINE or as a device extension)

RealSense-Customer-Engineering, 2019-01-10:
[Realsense Customer Engineering Team Comment]
Closing it for now. Please reopen if the original issue was not addressed.

dorodnic, 2019-01-12:
@RealSense-Customer-Engineering let's not close issues market `enhancement`, these should propagate to backlog as change-requests

ev-mp, 2020-01-07:
The capability to query the motion sensor has been addressed with https://github.com/IntelRealSense/librealsense/pull/5468
Added to v2.32.0

closed 2020-01-07 by ev-mp