realsenseai/librealsense #2901: D435.   start_streaming(...) failed. UVC device is already streaming!    xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range

xizong, 2018-12-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |  05.10.13.00 | 
| Operating System & Version |    Linux 16,04| 
| Kernel Version (Linux Only)    |  (4.16.0)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.17.0                         | 
| Language                            |  {c++}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
     Hi,everyone. when I run the command "roslaunch  realsense2_camera rs_camera.launch", error happens.And from roslaunch realsense2_camera, only "demo_pointcloud.launch" works properly. I hope someone can help me solve this issue. Thank you!

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [6811]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 31ac2a52-00fb-11e9-82a4-dc8b28039517
process[rosout-1]: started with pid [6824]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [6841]
process[camera/realsense2_camera-3]: started with pid [6842]
[ INFO] [1544941448.791179174]: Initializing nodelet with 12 worker threads.
[ INFO] [1544941448.828606927]: RealSense ROS v2.1.2
[ INFO] [1544941448.828631100]: Running with LibRealSense v2.17.0
[ INFO] [1544941449.123237168]: getParameters...
[ INFO] [1544941449.164742641]: setupDevice...
[ INFO] [1544941449.164763039]: JSON file is not provided
[ INFO] [1544941449.164779655]: ROS Node Namespace: camera
[ INFO] [1544941449.164796728]: Device Name: Intel RealSense D435
[ INFO] [1544941449.164807047]: Device Serial No: 830112070318
[ INFO] [1544941449.164815537]: Device FW version: 05.10.13.00
[ INFO] [1544941449.164823914]: Device Product ID: 0x0B07
[ INFO] [1544941449.164831191]: Enable PointCloud: Off
[ INFO] [1544941449.164838269]: Align Depth: Off
[ INFO] [1544941449.164845488]: Sync Mode: Off
[ INFO] [1544941449.164903510]: Device Sensors: 
[ INFO] [1544941449.164923850]: Stereo Module was found.
[ INFO] [1544941449.164934199]: RGB Camera was found.
[ INFO] [1544941449.164982870]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1544941449.164995837]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1544941449.165004018]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1544941449.165017579]: setupPublishers...
[ INFO] [1544941449.166832419]: Expected frequency for depth = 30.00000
[ INFO] [1544941449.185863519]: Expected frequency for infra1 = 30.00000
[ INFO] [1544941449.202491228]: Expected frequency for infra2 = 30.00000
[ INFO] [1544941449.219663576]: Expected frequency for color = 30.00000
[ INFO] [1544941449.236620822]: setupStreams...
 16/12 14:24:09,238 WARNING [140178048231296] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1544941449.239939411]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1544941449.240115329]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1544941449.240268252]: infra2 stream is enabled - width: 640, height: 480, fps: 30
 16/12 14:24:09,241 WARNING [140178048231296] (backend-v4l2.cpp:1248) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 16/12 14:24:09,242 WARNING [140178048231296] (sensor.cpp:338) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1544941449.246570658]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1544941449.257971991]: num_filters: 0
[ INFO] [1544941449.258000941]: publishStaticTransforms...
[ INFO] [1544941449.258166850]: RealSense Node Is Up!
[ INFO] [1544941449.258205804]: Start updating dynamic parameters...
 16/12 14:24:09,310 WARNING [140178048231296] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 16/12 14:24:09,310 WARNING [140178048231296] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 16/12 14:24:09,310 WARNING [140178048231296] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 16/12 14:24:09,310 WARNING [140178048231296] (types.cpp:57) start_streaming(...) failed. UVC device is already streaming!
 16/12 14:24:09,376 ERROR [140178048231296] (types.h:188) xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range
[ERROR] [1544941449.376423759]: Failed. Skip initialization of parameter 24
[ INFO] [1544941449.376899586]: Done updating dynamic parameters...
 05.10.13.00

gopikaajaykumar, 2018-12-21:
I am receiving the same error messages when using the D415. I am using the latest firmware and librealsense2 versions.

RealSense-Customer-Engineering, 2019-01-11:
[Realsense Customer Engineering Team Comment]
Hi xizong,

Do you have any update on this one?

I checked out LibRealSense SDK 2.17.0 source code and made a new build.
And then checked out ROS wrapper 2.1.1, which requires LRS SDK 2.17.0 (https://github.com/intel-ros/realsense).
After followed the steps, I tested a few launches, including the one you mentioned. No issue from my tests.

    roslaunch realsense2_camera rs_camera.launch

This is done off an Intel NUC w/ Ubuntu 16.04 (kernel 4.13)

Thanks!

RealSenseCustomerSupport, 2019-01-25:
Hi xizong,

Wonder if there is any updates on this?
If nothing else, we are going to close it.

Thank you!

closed 2019-02-11 by RealSenseCustomerSupport