realsenseai/librealsense #2904: operation(draw lines and circles) on pointcloud

guantong1, 2018-12-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300  | 
| Firmware Version                   |3.21.0.0 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.14.1 }                          | 
| Language                            | opencv/pcl/python }                          | 
| Segment			|  others }                          | 

### Issue Description
    i am fingding​ a way to draw lines and circles on pointcloud and display the pointcloud in real time like a video
    In pcl, there is a function that can draw line segments between two points. The method of drawing lines between points is very useful.  It is a member of the visual function pcl::visualizeton. i cannot find this in python-pcl.
    i detected feagure points in religned RGB-D images , i want to showthese feagure points in pointclouds.
    i got the 3d feagure coordinates by deprojected depth pixels. i also got the pointclouds :points = pc.calculate(depth_frame).

RealSense-Customer-Engineering, 2018-12-20:
[Realsense Customer Engineering Team Comment]
Hi @guantong1,

You can review below pointcloud examples to see how it works.
https://github.com/IntelRealSense/librealsense/blob/bb6643c4f506fe91daa8a77b8a469c29bf863dbe/wrappers/python/examples/readme.md

dorodnic, 2019-02-04:
Hopefully the response was useful. Feel free to open a new issue if not fully resolved

closed 2019-02-04 by dorodnic