realsenseai/librealsense #2905: Core Dump trying to get intrinsics for motion stream

barnjamin, 2018-12-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.10.13.00  | 
| Operating System & Version |  Ubuntu 18 | 
| Kernel Version (Linux Only)    |  4.15.0-42    | 
| Platform                                 | PC  |
| SDK Version                            | `development` branch - 8446e160125d8d723830568b460e4a23fe0f6c89                   | 
| Language                            |  C++                   | 

### Issue Description
Core dump trying to get motion intrinsics

Relevant section
```
    ...
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_ACCEL);
    cfg.enable_stream(RS2_STREAM_GYRO);
    rs2::pipeline_profile profile = pipe.start(cfg);

    auto accel_stream = profile.get_stream(RS2_STREAM_ACCEL).as<rs2::motion_stream_profile>();
    auto gyro_stream = profile.get_stream(RS2_STREAM_GYRO).as<rs2::motion_stream_profile>();

    std::cout << "Accel FPS: " << accel_stream.fps() << std::endl;
    std::cout << "Gyro FPS: " << gyro_stream.fps() << std::endl;

    auto accel_intr = accel_stream.get_motion_intrinsics();
    auto gyro_intr = gyro_stream.get_motion_intrinsics();
    ...
```

Output

```
Accel FPS: 63
Gyro FPS: 200
terminate called without an active exception
Aborted (core dumped)
```

Commenting out the calls to get intrinsics, the program continues without issue.

barnjamin, 2018-12-28:
Also just tried getting intrinsics from rs-sensor-control and got the same error

```
Sensor consists of 2 streams: 
  - Gyro #0
  - Accel #0
Sensor provides the following stream profiles:
0  : Gyro #0
1  : Gyro #0
2  : Accel #0
3  : Accel #0

Please select the desired streaming profile: 2

Failed to get intrinsics for the given stream. Depth Module V2 intrinsic invalidated : Accel !
Show stream intrinsics again?[y/n]: y

======================================================
```

barnjamin, 2018-12-28:
using the current latest release (2.17.0) I get:

```
RealSense error calling rs2_get_motion_intrinsics(mode:0x557f5f9bfd48, intrinsics:0x7fffea9c3250):
    OpCodes do not match! Sent 79 but received -21!
```

dorodnic, 2018-12-29:
Hi @barnjamin 
Sorry for delayed response. The D435i does not have pre-calibrated IMU intrinsics, but we are working right now to provide a simple manual calibration script and adjusting librealsense to take advantage of this data when available.

barnjamin, 2018-12-29:
ah, great thanks!

RealSense-Customer-Engineering, 2019-01-08:
[Realsense Customer Engineering Team Comment]
Hi barnjamin,

This is an ongoing work.
Please check back to the github later or subscribe to RS updates.

Thank you!