realsenseai/librealsense #2910: the depth max value is 0.001 * 255

jinfagang, 2018-12-17:
I got a real distance scale from camera, and it is 0.001, I think it means every pixel accordingly real distance in meters, as the pixel max value is 255, so the max distance is 0.001 * 255 = 0.255m

That make no sense............... How should I get the **real** *real* distance from depth???

dorodnic, 2018-12-17:
Hi @jinfagang 
Every depth pixel is two bytes (`uint16_t`)

jinfagang, 2018-12-18:
@dorodnic  Hi, what do u mean by `uint16_t`, I just got the depth image from ROS published topic. So where to convert the pixel value into real distance (if it is only mutiply the scale, then the max pixel value is 255, the max value distance is limited)............... Here is my code:

```

void realsense_depth_callback(const sensor_msgs::ImageConstPtr &msg) {
    cv_bridge::CvImagePtr cv_ptr;
    cv::Mat img;
    cv_ptr = cv_bridge::toCvCopy(msg);
    crt_depth = cv_ptr->image;
//    cv::rectangle(crt_depth, p1, p2, cv::Scalar(12, 242, 35));
    // convert to CV_8UC1
    cv::Mat c1_depth;
    crt_depth.convertTo(c1_depth, CV_8UC3);

    uint8_t* pixel_ptr = crt_depth.data;

    if (!crt_img_color.empty()) {
        cv::Mat color_tmp;
        crt_img_color.copyTo(color_tmp);

        // if distance > 1m, wipe it from crt_image
        for (int j = 0; j < crt_depth.rows; ++j) {
            for (int k = 0; k < crt_depth.cols; ++k) {
                uint8_t d = pixel_ptr[j*crt_depth.cols + k];

                // in meters
                float real_dist = d * rs_real_distance_scale;
                if (real_dist > 0.5) {
                    // set i,j pixel value in color image to black
                    color_tmp.at<cv::Vec3b>(j, k)[0] = 0;
                    color_tmp.at<cv::Vec3b>(j, k)[1] = 255;
                    color_tmp.at<cv::Vec3b>(j, k)[2] = 0;
                }
            }
        }
        cout << color_tmp.rows << "x" << color_tmp.cols << endl;
        // convert depth to color for better visualization
        cv::Mat bgr;
        cv::applyColorMap(c1_depth, bgr, cv::COLORMAP_JET);
//        cv::imshow("depth", crt_depth);
        cv::imshow("depth", bgr);
        cv::imshow("wiped color", color_tmp);
        cv::waitKey(1);
    }
}

```

jinfagang, 2018-12-18:
BTW, when I using `uint16_t* pixel_ptr = crt_depth.data;`, build error..........

RealSense-Customer-Engineering, 2018-12-20:
[Realsense Customer Engineering Team Comment]
@jinfagang
crt_depth should be CV_16UC1 format. Please refer to the link below.
https://github.com/intel-ros/realsense/blob/2.1.2/realsense2_camera/src/base_realsense_node.cpp#L29

So by default you will get 16  bit depth with a depth unit of 0.001(stands for 1mm). This means the max
range will be ~65m.

For how to get the real distance, please refer to below links.
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/cv-helpers.hpp#L42

RealSense-Customer-Engineering, 2019-01-03:
[Realsense Customer Engineering Team Comment]
@jingfagang Any other questions about this ticket? Please update. Thanks!

closed 2019-01-08 by jinfagang

RealSense-Customer-Engineering, 2019-01-11:
[Realsense Customer Engineering Team Comment]
@jingfagang Close the ticket at this point. If you still have any other questions, please feel free to open another ticket. Thanks!